wled/WLED #3535: WLED doesn't sync after DDP has been off, until WLED is rebooted

kristofferR, 2023-11-19:
### What happened?

I've got a WLED&Hyperion setup on my TV. When I turn off the TV and Hyperion stops sending data, the sync stops as expected. However, when I turn on the TV again and Hyperion starts sending new data, WLED doesn't sync. However, as soon as I reboot WLED it works properly again, every time.

Before updating to v0.14, everything worked fine. I used to be able to click the Sync button in WLED to stop and start syncing, but in v0.14 the sync button doesn't seem to do anything.

I have filmed a video here that shows the bug in action:

https://github.com/Aircoookie/WLED/assets/481270/5764be7c-0d22-46eb-9115-b73272d1a0b1

### To Reproduce Bug

Have DDP sync enabled. Sync something, then turn off the sync at source, then start syncing again.

### Expected Behavior

WLED shouldn't have to be rebooted to get DDP sync working again.

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.14.0 (build 2310130)

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2023-11-19:
Sync has nothing to do with DDP. Leave it off.

kristofferR, 2023-11-19:
> Sync has nothing to do with DDP. Leave it off.

I've updated the language to be more clear.

Yuke-hd, 2024-02-12:
I'm experiencing the same issue.

After stopping DDP streaming source (shut down Hyperion) the WLED webui is still responsive but even the built in effects won't work

[image]

See above screenshot: I was trying to show a solid red but the leds were off and PEAK is showing nothing

Yuke-hd, 2024-02-12:
https://github.com/Aircoookie/WLED/issues/3700#issuecomment-1914889050

might be related

kristofferR, 2024-02-12:
> [#3700 (comment)](https://github.com/Aircoookie/WLED/issues/3700#issuecomment-1914889050)
> 
> might be related

Awesome, yeah, upgrading to 0.15 nightly fixed it for me.

closed 2024-06-19 by github-actions[bot]