wled/WLED #3538: Add support for XIAO ESP32-S3 and many other ESP32-based new boards in espressif32

BADA55A1, 2023-11-21:
**Is your feature request related to a problem? Please describe.**
1. I've got XIAO ESP32-S3 and neither release `WLED_0.14.0_ESP32-S3_8MB.bin` or custom compiled from main branch `esp32s3dev_8MB.bin` worked properly: after flash board was power cycling with 'entry 0x403c98d8' or similar error.

2. This board appears in platformio boards with name "seeed_xiao_esp32s3"
    ```bash
    $ platformio boards | grep xiao
    seeed_xiao                     SAMD21G18A   48MHz        256KB    32KB   Seeeduino XIAO
    seeed_xiao_esp32c3                   ESP32C3  160MHz       4MB      320KB   Seeed Studio XIAO ESP32C3
    seeed_xiao_esp32s3                   ESP32S3  240MHz       8MB      320KB   Seeed Studio XIAO ESP32S3
    ``` 
    But if I set `board = seeed_xiao_esp32s3` for some configuration in `platformio.ini`, I'm getting error:
    ```
    Processing esp32s3dev_8MB (board: seeed_xiao_esp32s3; platform: espressif32@5.3.0; framework: arduino)
    UnknownBoard: Unknown board ID 'seeed_xiao_esp32s3'
    ```
    
    Probably this error appears due to older espressif32 in `[esp32]` section used:
    ```
    ...
    [esp32]
    #platform = https://github.com/tasmota/platform-espressif32/releases/download/v2.0.2.3/platform-espressif32-2.0.2.3.zip
    platform = espressif32@3.5.0
    ...
    ```

    I'm fixing the error above using most recent platform-espressif32:
    ```
    ...
    [env:esp32s3dev_8MB]
    ;; ESP32-S3-DevKitC-1 development board, with 8MB FLASH, no PSRAM (flash_mode: qio)
    board = seeed_xiao_esp32s3
    ; platform = ${esp32s3.platform}
    platform = https://github.com/platformio/platform-espressif32.git
    platform_packages = ${esp32s3.platform_packages}
    ...
    ```
    Firmware built using settings above works on my XIAO ESP32S3 without issues.  

_Am I lucky with this particular board to make it work in such an easy way?_
_Or running WLED on new exotic ESP32-based boards is always a lottery?_ 

**Describe the solution you'd like**
Is it possible to extend support for any board supported by latest espressif32?

**Describe alternatives you've considered**
Maybe provide some dev-template with an experimental firmware build recipe for new boards with latest espressif32?

**Additional context**

[My reply](https://wled.discourse.group/t/installation-on-xiao-esp32s3/9965/2?u=bada55a1) on [Installation on Xiao ESP32S3 forum thread](https://wled.discourse.group/t/installation-on-xiao-esp32s3/9965)
My `platformio.ini` [snippet](https://gist.github.com/BADA55A1/2b4d7cf2e1a86a7659ba2dba4a0425c7) 

---

I just started my adventure with WLED and ESP32 and right now I'm too far from understanding project code base to prepare a pull request.

I'm already in love with WLED project (had built some decorations with SK6812 and WLED with audioreactive mod, 2d animations for a party 3 months ago and it worked beautifully).
Thank you for such great project! ❤️

blazoncek, 2023-11-22:
Although many users successfully use ESP32-S3 (or C3, S2) all of them are still in experimental, POC stage.
If you've found a solution for a particular board then please provide a complete PIO environment and possibly make a PR.
Unfortunately we rely on community to do extensive board support as we do not own each and every board possible.

FYI we will not include unreleased platform (i.e. https://github.com/platformio/platform-espressif32.git) in regular publicly available builds at any time as there may have breaking features included. We prefer a known release (i.e. espressif32@5.2.0).

softhack007, 2023-11-22:
Fully agreed with what @blazoncek explained. Additionally, many board defs in platformIO are only marginally different from the generic definitions. 

In your case I only saw two main differences:
* the board is a standard PSRAM variant that needs qpi_opi mode 
* you need USB_CDC for debugging
* the board seems to use the "tinyUF2" bootloader which is not supported by WLED.

Actually you could try to build WLED using [`env:esp32s3dev_8MB_PSRAM_opi`](https://github.com/Aircoookie/WLED/blob/32e724e744828498ba03703ed01e4179592d7aa7/platformio.ini#L507)

We use platform versions 5.3.0 for -S3  right now, because espressif regularly breaks things;so we prefer to stick with known-good  versions. As example: in the latest arduino-esp32 (platform > 6.3.1) audio support is broken, and you either have no audio at all, or lots of static noise that makes audioreactive go gaga. 

You can always add your own experimetal build environment in platdormio_override.ini.

BADA55A1, 2023-11-22:
> Actually you could try to build WLED using [`env:esp32s3dev_8MB_PSRAM_opi`](https://github.com/Aircoookie/WLED/blob/32e724e744828498ba03703ed01e4179592d7aa7/platformio.ini#L507). Please upload from platformIO (build & upload), to make sure that all needed parts (bootloader, partitions, app0, wled) are properly uploaded to the board.

Ok, I'm definitely going to try this option later.

---

> We use platform version 5.3.0 for -S3 right now, because espressif regularly breaks things; so we prefer to stick with known-good versions. As example: in the latest arduino-esp32 (platform > 6.3.1) audio support is broken, and you either have no audio at all, or lots of static noise that makes audioreactive go gaga.

This applies to analog audio only? I'm using I2S mic.

softhack007, 2023-11-22:
> This applies to analog audio only? I'm using I2S mic.

Yes I was talking about digital I2S audio (mic or line-in). But it was just an example - there are more regressions in latest espressif frameworks, and it's always a suprise to find out what's  newly broken...

blazoncek, 2024-04-24:
There are 3 available S3 builds (actually 2 matter OPI & QSPI) since 0.15.0-b3.

closed 2024-04-24 by blazoncek