wled/WLED #3541: Playlist transition time being ignored/reset

KonkaBass, 2023-11-24:
### What happened?

I have a playlist, that I'm attempting to blend between presets seamlessly.
`{"playlist":{"ps":[7,9,8,10,11,13,14,12],"dur":[3000,3000,3000,3000,3000,3000,3000,3000],"transition":[600,600,600,600,600,600,600,600],"repeat":0,"end":0,"r":1},"on":true}`
aka 5mins static with a 1min blend

The playlist was ignoring the playlist transition time and just using the default transition time. I couldn't understand why, I solved this temporarily when I opened WLED on desktop instead of mobile and saw all controls at once, this led me to change the Transition time under my two Segments after launching the Playlist.

This worked, allowing the playlist to perform my smooth (60s) transition. But it's temporary, because if I select any preset, (pretty much all of which have a default 0.7 transition time), the segment transition resets to 0.7 and if I select the Playlist again, the original issue returns, where the Playlist transition time is ignored, and the segment transition time stays at 0.7, and the transitions are sudden.

### To Reproduce Bug

Have more than 1 segment. Use presets with the default transition time in a Playlist. Set transition time to anything longer.

### Expected Behavior

If a transition time is entered into a Playlist, that is the transition time that is used across all segments, regardless of transition time used in Presets used inside Playlist.

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.14.0 (build 2310130)

### Which microcontroller/board are you seeing the problem on?

ESP32

### Relevant log/trace output

```shell
{"0":{},"1":{"on":true,"bri":132,"transition":7,"mainseg":0,"seg":[{"id":0,"start":0,"stop":1,"grp":1,"spc":0,"of":0,"on":true,"frz":false,"bri":255,"cct":46,"set":0,"col":[[0,0,0,255],[0,0,0,0],[0,0,0,0]],"fx":0,"sx":123,"ix":128,"pal":0,"c1":128,"c2":128,"c3":16,"sel":true,"rev":false,"mi":false,"o1":false,"o2":false,"o3":false,"si":0,"m12":0},{"id":1,"start":1,"stop":2,"grp":1,"spc":0,"of":0,"on":true,"frz":false,"bri":255,"cct":46,"set":0,"col":[[0,0,0,255],[0,0,0,0],[0,0,0,0]],"fx":0,"sx":128,"ix":128,"pal":0,"c1":128,"c2":128,"c3":16,"sel":true,"rev":false,"mi":false,"o1":false,"o2":false,"o3":false,"si":0,"m12":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0}],"n":"3000K"},"6":{"on":true,"bri":130,"transition":7,"mainseg":0,"seg":[{"id":0,"start":0,"stop":1,"grp":1,"spc":0,"of":0,"on":true,"frz":false,"bri":255,"cct":46,"set":0,"col":[[0,225,255,0],[0,0,0,0],[0,0,0,0]],"fx":0,"sx":123,"ix":128,"pal":0,"c1":128,"c2":128,"c3":16,"sel":true,"rev":false,"mi":false,"o1":false,"o2":false,"o3":false,"si":0,"m12":0},{"id":1,"start":1,"stop":2,"grp":1,"spc":0,"of":0,"on":true,"frz":false,"bri":255,"cct":46,"set":0,"col":[[0,225,255,0],[0,0,0,0],[0,0,0,0]],"fx":0,"sx":128,"ix":128,"pal":0,"c1":128,"c2":128,"c3":16,"sel":true,"rev":false,"mi":false,"o1":false,"o2":false,"o3":false,"si":0,"m12":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0}],"n":"Blue Sky"},"3":{"on":true,"bri":138,"transition":7,"mainseg":0,"seg":[{"id":0,"start":0,"stop":1,"grp":1,"spc":0,"of":0,"on":true,"frz":false,"bri":255,"cct":255,"set":0,"col":[[1,1,1,255],[0,0,0,0],[0,0,0,0]],"fx":0,"sx":123,"ix":128,"pal":0,"c1":128,"c2":128,"c3":16,"sel":true,"rev":false,"mi":false,"o1":false,"o2":false,"o3":false,"si":0,"m12":0},{"id":1,"start":1,"stop":2,"grp":1,"spc":0,"of":0,"on":true,"frz":false,"bri":255,"cct":255,"set":0,"col":[[1,1,1,255],[0,0,0,0],[0,0,0,0]],"fx":0,"sx":128,"ix":128,"pal":0,"c1":128,"c2":128,"c3":16,"sel":true,"rev":false,"mi":false,"o1":false,"o2":false,"o3":false,"si":0,"m12":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0}],"n":"5600K"}             ,"2":{"on":true,"bri":135,"transition":7,"mainseg":0,"seg":[{"id":0,"start":0,"stop":1,"grp":1,"spc":0,"of":0,"on":true,"frz":false,"bri":255,"cct":142,"set":0,"col":[[1,1,1,255],[0,0,0,0],[0,0,0,0]],"fx":0,"sx":123,"ix":128,"pal":0,"c1":128,"c2":128,"c3":16,"sel":true,"rev":false,"mi":false,"o1":false,"o2":false,"o3":false,"si":0,"m12":0},{"id":1,"start":1,"stop":2,"grp":1,"spc":0,"of":0,"on":true,"frz":false,"bri":255,"cct":142,"set":0,"col":[[1,1,1,255],[0,0,0,0],[0,0,0,0]],"fx":0,"sx":128,"ix":128,"pal":0,"c1":128,"c2":128,"c3":16,"sel":true,"rev":false,"mi":false,"o1":false,"o2":false,"o3":false,"si":0,"m12":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0}],"n":"4000K"},"4":{"on":true,"bri":255,"transition":7,"mainseg":0,"seg":[{"id":0,"start":0,"stop":1,"grp":1,"spc":0,"of":0,"on":true,"frz":false,"bri":255,"cct":46,"set":0,"col":[[0,17,179,255],[0,0,0,0],[0,0,0,0]],"fx":0,"sx":123,"ix":128,"pal":0,"c1":128,"c2":128,"c3":16,"sel":true,"rev":false,"mi":false,"o1":false,"o2":false,"o3":false,"si":0,"m12":0},{"id":1,"start":1,"stop":2,"grp":1,"spc":0,"of":0,"on":true,"frz":false,"bri":255,"cct":46,"set":0,"col":[[0,17,179,255],[0,0,0,0],[0,0,0,0]],"fx":0,"sx":128,"ix":128,"pal":0,"c1":128,"c2":128,"c3":16,"sel":true,"rev":false,"mi":false,"o1":false,"o2":false,"o3":false,"si":0,"m12":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0}],"n":"Daylight 8700K"},"7":{"mainseg":0,"seg":[{"id":0,"start":0,"stop":1,"grp":1,"spc":0,"of":0,"on":true,"frz":false,"bri":255,"cct":121,"set":0,"col":[[157,0,255,0],[0,0,0,0],[0,0,0,0]],"fx":0,"sx":128,"ix":128,"pal":0,"c1":128,"c2":128,"c3":16,"sel":true,"rev":false,"mi":false,"o1":false,"o2":false,"o3":false,"si":0,"m12":1},{"id":1,"start":1,"stop":2,"grp":1,"spc":0,"of":0,"on":true,"frz":false,"bri":255,"cct":46,"set":0,"col":[[77,255,0,0],[0,0,0,0],[0,0,0,0]],"fx":0,"sx":128,"ix":128,"pal":0,"c1":128,"c2":128,"c3":16,"sel":true,"rev":false,"mi":false,"o1":false,"o2":false,"o3":false,"si":0,"m12":1},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0}],"n":"Split Purp/Green"} ,"8":{"mainseg":0,"seg":[{"id":0,"start":0,"stop":1,"grp":1,"spc":0,"of":0,"on":true,"frz":false,"bri":255,"cct":121,"set":0,"col":[[0,200,255,0],[0,0,0,0],[0,0,0,0]],"fx":0,"sx":128,"ix":128,"pal":0,"c1":128,"c2":128,"c3":16,"sel":true,"rev":false,"mi":false,"o1":false,"o2":false,"o3":false,"si":0,"m12":1},{"id":1,"start":1,"stop":2,"grp":1,"spc":0,"of":0,"on":true,"frz":false,"bri":255,"cct":46,"set":0,"col":[[255,0,217,0],[0,0,0,0],[0,0,0,0]],"fx":0,"sx":128,"ix":128,"pal":0,"c1":128,"c2":128,"c3":16,"sel":true,"rev":false,"mi":false,"o1":false,"o2":false,"o3":false,"si":0,"m12":1},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0}],"n":"Split Pink/Teal"} ,"9":{"mainseg":0,"seg":[{"id":0,"start":0,"stop":1,"grp":1,"spc":0,"of":0,"on":true,"frz":false,"bri":255,"cct":121,"set":0,"col [... cut ...] ,"mi":false,"o1":false,"o2":false,"o3":false,"si":0,"m12":1},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0}],"n":"Split Purple/Orange"}}
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2023-11-25:
I could not reproduce.
Please post your config as well (cfg.json), please.

harueg, 2023-11-26:
Hi
Having the same issue. Transition time in playlist 3s. General transition time 0,7s.
Works fine with general transition time of 3s. It seems as general transition is only applied after reset.
[wled_cfg_Testschaukel.json](https://github.com/Aircoookie/WLED/files/13467804/wled_cfg_Testschaukel.json)
[wled_presets_Testschaukel.json](https://github.com/Aircoookie/WLED/files/13467805/wled_presets_Testschaukel.json)

blazoncek, 2023-11-26:
Ok, this one is weird.
I was not able to reproduce using your "Rottöne" playlist *until* I changed global transition time during playlist play.
After that change the playlist transition time was no longer obeyed until I turned WLED off reset global transition time and started playlist again.

harueg, 2023-11-26:
As long as the playlist transition time is shorter than the global transition time it works.
GTT seems to limit PTT.

KonkaBass, 2023-11-26:
here's the config as requested
[wled_cfg_WLED An-Deca.json](https://github.com/Aircoookie/WLED/files/13468369/wled_cfg_WLED.An-Deca.json)
Also for the sake of testing I've made a playlist with a much shorter transition time, (plus my hack attempt I write about below)
[wled_presets_WLED An-Deca.json](https://github.com/Aircoookie/WLED/files/13468436/wled_presets_WLED.An-Deca.json)

some things I've noticed / alterations to the issue I've encountering since my initial report.

On mobile (android 1.1.0) and on web UI
If I want to get a long transition I must edit global transition time **before** I engage the playlist.
If I try and edit global time after the playlist has started, the global transition time will reset to the saved transition time of the Preset used in the playlist once the duration time of the current preset has concluded.

I tried seeing if I could hack the behavior into behaving by saving some presets with a 5s transition and making a separate playlist using those presets.

With a more testing. I've come to find that global transition time is tied to if "include brightness" is checked during preset creation meaning a brightness included preset will change the global transition value, and any preset used after that _doesn't_ have brightness included will follow that new global transition time. **Also**, even if a playlist contains presets with brightness included, the playlist will ignore that and follow the global transition time set by a previous brightness included playlist.

and while I know it doesn't exactly pertain to the issue here, now that I know this: Does anyone think that global transition time should be tied to "save current state" and not "include brightness"? I personally could see more situations where I'd want to maintain a brightness value separately from a transition time. Anyways, more of a side thought...

Issue is still GTT is being used instead of PTT.

blazoncek, 2023-11-27:
This is so f***ed up!
We may need @Aircoookie insights on `transitionDelayTemp` as that's his invention to keep legacy compatibility (temporary transition, transition from notifications, etc).

@KonkaBass there are quite a few legacy decisions that make current development difficult to say the least. Changing them is even greater PITA.

closed 2023-12-18 by blazoncek