wled/WLED #3543: WLED powers on the lights when "On" key and "false" value are provided via JSON

almostengr, 2023-11-24:
### What happened?

When making a POST request to the "/json/state" endpoint with the below JSON,the power to the lights turn on when they should not. 

```
{
    "On": false,
    "bri" :255
}
```

In the above JSON, the "O" in "On" is uppercase.  The same thing happens if "ON" is use with false value. 

### To Reproduce Bug

Video of the bug https://youtu.be/HwQEIcyD7HI

- Turn off the WLED lights using the Power button on the WLED page
- open API client (e.g. Postman) 
- make API call to the "/json/state" with the "on" key in Sentence or Upper Case. Example JSON below. 
```json
{
"ON": false,
"brI": 255
}
```

- submit the above request as POST
- Lights will turn on

### Expected Behavior

Expectation 1) 
Light should be powered off instead of On. 

Expectation 2)
The request should fail to complete, and the JSON response should be 

```
{
    "success": false
}
```

### Install Method

Binary from WLED.me

### What version of WLED?

0.13.3

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2023-11-24:
The valid JSON property for on state is "on", not "On" or "ON".
Please consult https://kno.wled.ge

closed 2023-11-24 by blazoncek

almostengr, 2023-11-24:
> The valid JSON property for on state is "on", not "On" or "ON". Please consult https://kno.wled.ge

@blazoncek This is still and issue because even if "On" or "ON" is sent, the response returns ```"success" : true``` which is mislead. The request should ```"success": false``` in this scenario as an invalid key was passed to it. 

If this is not going to be fixed, then the documentation needs to be updated to state that ```true``` will still be returned even when the wrong keys are passed to the end point.

blazoncek, 2023-11-25:
JSON is successfully parsed nevertheless. Unknown entries are ignored.

blazoncek, 2023-11-25:
BTW you are welcome to update the documentation if you think it deserves updating or clarification.
Fork and PR [WLED-Docs](https://github.com/Aircoookie/WLED-Docs).