wled/WLED #3549: Multiple Potentiometers not working since 0.14.0.b1

rhinomods, 2023-11-25:
### What happened?

I had 0.13.3 working with multiple potentiometers for  brightness, speed, intensity, and color working, Updated to 0.14.0 and only the first enabled button would work. (brightness.)

### To Reproduce Bug

with Esp32dev v1, flashed with wled 0.14.0
I attached 10k Potentiometers to ADC 0,3,6,7 (gpio 36,39,34,35)
Under [LED PREFERENCES] set buttons 0-3 to analog inverted and (Respective GPIO pin#)
Under [TIME AND MACROS] set buttonActions for Button 0-3 SHORT 0,0,0,0 LONG 0,0,0,0 DOUBLE to 200,250,249,248

Adjust Potentiometers, I notice ONLY the First button that is enabled or set to ANALOG inverted will work, the others will not register. 
IF I set button 0 to DISABLED, button 1 will work but others will not. ETC.

I attempted to use the checkbox to enable or disable pull up resistor No change.

### Expected Behavior

On 0.13.3 I was able to have multiple Potentiomers working.

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.14.0

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

softhack007, 2023-12-17:
@rhinomods any new insights from using the debug version?

Hubes101, 2024-01-16:
I just updated my esp32 and I’m having the same problem
I was wondering if this has been fixed and I just haven’t noticed it yet

softhack007, 2024-01-16:
> I was wondering if this has been fixed and I just haven’t noticed it yet

As far as I know, there was no progress on this issue.
@blazoncek asked @rhinomods for help on debugging, but no response or feedback arrived. At least that's my understanding.

blazoncek, 2024-01-16:
There was no feedback. Last communication was on December 16th to say "working on it. sorry for long delay. hopefully i can respond tomorrow".
I did check the code but couldn't find any clue where the problem may be.

blazoncek, 2024-02-10:
will be available in 0.14.2 and/or 0.15

closed 2024-02-10 by blazoncek