wled/WLED #3550: WLED 0.14.0 effects are much faster than on 0.13.3

H0nzik1, 2023-11-26:
### What happened?

After update to 0.14.0 I cannot make effects as slow as on 0.13.3

### To Reproduce Bug

Just compare 0.13.3 vs 0.14.0 for example on Tetrix effect on speed 10. I cannot make it slow on 0.14.0 as it was on 0.13.3

### Expected Behavior

Expected behavior is that effects are much slower. Between 128-256 they are unusable. I tested mainly Tetrix effect but others seems to be too fast too. 

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.14.0

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2023-11-26:
Hi, unfortunately that was the intended change.
The falling speed is now time based, from 250ms to 5s (the time it takes for a brick to fall the whole length).

508loopdetect, 2023-11-30:
I also notice too much animation speed on almost every pattern, this is annoying fast.  does not look smotth any more

i wish there is a slow down to 0.001 %  or a 1/10 divider button somewhere,  cant see 1% 

why is here a wontfix added, he just made tetrix as an example?

blazoncek, 2023-11-30:
Please enumerate *all* effects and their usable speeds. And post them in a table with desired % of increase or decrease of minimum and maximum speeds.

Only after that we can discuss *if* there are going to be any changes. And please have people upvote this issue.

H0nzik1, 2023-12-02:
> Please enumerate _all_ effects and their usable speeds. And post them in a table with desired % of increase or decrease of minimum and maximum speeds.
> 
> Only after that we can discuss _if_ there are going to be any changes. And please have people upvote this issue.

Usable speeds are relative to what people like. But even at lowest speed I think many of them are barely usable now. Just add more spread to speed setting?  Also there is bug when you set speed 0 which is somehow faster than numbers like speed 1 or 10.

I propose slower all effects 10x than they are now. Even in WLED 0.13.3 they are quite fast on 60LED/1m strips. I have set effects to speeds around 60 out of 256 on 0.13.3, and I guess 0.14.0 is now about 5x faster than 0.13.3, thats why I think 10x slower will be usable at speed 128 out of 256. 

ID | Name | Descrease speed to % of current speed
186 | Akemi | 10%
27 | Android | 10%
38 | Aurora | 10%
183 | Black Hole | 10%
115 | Blends | 10%
1 | Blink | 10%
26 | Blink Rainbow | 10%
121 | Blobs | 10%
163 | Blurz | 10%
91 | Bouncing Balls | 10%
68 | Bpm | 10%
2 | Breathe | 10%
88 | Candle | 10%
102 | Candle Multi | 10%
28 | Chase | 10%
37 | Chase 2 | 10%
54 | Chase 3 | 10%
31 | Chase Flash | 10%
32 | Chase Flash Rnd | 10%
30 | Chase Rainbow | 10%
29 | Chase Random | 10%
111 | Chunchun | 10%
167 | Colored Bursts | 10%
34 | Colorful | 10%
8 | Colorloop | 10%
74 | Colortwinkles | 10%
67 | Colorwaves | 10%
119 | Crazy Bees | 10%
159 | DJ Light | 10%
152 | DNA | 10%
182 | DNA Spiral | 10%
112 | Dancing Shadows | 10%
18 | Dissolve | 10%
19 | Dissolve Rnd | 10%
124 | Distortion Waves | 10%
164 | Drift | 10%
123 | Drift Rose | 10%
96 | Drip | 10%
7 | Dynamic | 10%
117 | Dynamic Smooth | 10%
12 | Fade | 10%
49 | Fairy | 10%
51 | Fairytwinkle | 10%
69 | Fill Noise | 10%
66 | Fire 2012 | 10%
45 | Fire Flicker | 10%
149 | Firenoise | 10%
42 | Fireworks | 10%
90 | Fireworks 1D | 10%
89 | Fireworks Starburst | 10%
110 | Flow | 10%
179 | Flow Stripe | 10%
155 | Freqmap | 10%
138 | Freqmatrix | 10%
141 | Freqpixels | 10%
137 | Freqwave | 10%
177 | Frizzles | 10%
160 | Funky Plank | 10%
139 | GEQ | 10%
172 | Game Of Life | 10%
120 | Ghost Rider | 10%
87 | Glitter | 10%
46 | Gradient | 10%
156 | Gravcenter | 10%
157 | Gravcentric | 10%
158 | Gravfreq | 10%
132 | Gravimeter | 10%
82 | Halloween Eyes | 10%
100 | Heartbeat | 10%
180 | Hiphotic | 10%
58 | ICU | 10%
64 | Juggle | 10%
130 | Juggles | 10%
168 | Julia | 10%
75 | Lake | 10%
41 | Lighthouse | 10%
57 | Lightning | 10%
176 | Lissajous | 10%
47 | Loading | 10%
131 | Matripix | 10%
153 | Matrix | 10%
154 | Metaballs | 10%
76 | Meteor | 10%
77 | Meteor Smooth | 10%
135 | Midnoise | 10%
59 | Multi Comet | 10%
70 | Noise 1 | 10%
71 | Noise 2 | 10%
72 | Noise 3 | 10%
73 | Noise 4 | 10%
107 | Noise Pal | 10%
146 | Noise2D | 10%
143 | Noisefire | 10%
136 | Noisemeter | 10%
145 | Noisemove | 10%
126 | Octopus | 10%
62 | Oscillate | 10%
101 | Pacifica | 10%
65 | Palette | 10%
98 | Percent | 10%
147 | Perlin Move | 10%
105 | Phased | 10%
109 | Phased Noise | 10%
128 | Pixels | 10%
129 | Pixelwave | 10%
97 | Plasma | 10%
178 | Plasma Ball | 10%
133 | Plasmoid | 10%
174 | Polar Lights | 10%
95 | Popcorn | 10%
63 | Pride 2015 | 10%
144 | Puddlepeak | 10%
134 | Puddles | 10%
162 | Pulser | 10%
78 | Railway | 10%
43 | Rain | 10%
9 | Rainbow | 10%
33 | Rainbow Runner | 10%
5 | Random Colors | 10%
79 | Ripple | 10%
148 | Ripple Peak | 10%
99 | Ripple Rainbow | 10%
185 | Rocktaves | 10%
15 | Running | 10%
52 | Running Dual | 10%
16 | Saw | 10%
10 | Scan | 10%
11 | Scan Dual | 10%
40 | Scanner | 10%
60 | Scanner Dual | 10%
122 | Scrolling Text | 10%
181 | Sindots | 10%
108 | Sine | 10%
92 | Sinelon | 10%
93 | Sinelon Dual | 10%
94 | Sinelon Rainbow | 10%
125 | Soap | 10%
0 | Solid | 10%
103 | Solid Glitter | 10%
83 | Solid Pattern | 10%
84 | Solid Pattern Tri | 10%
118 | Spaceships | 10%
20 | Sparkle | 10%
21 | Sparkle Dark | 10%
22 | Sparkle+ | 10%
85 | Spots | 10%
86 | Spots Fade | 10%
150 | Squared Swirl | 10%
39 | Stream | 10%
61 | Stream 2 | 10%
23 | Strobe | 10%
25 | Strobe Mega | 10%
24 | Strobe Rainbow | 10%
166 | Sun Radiation | 10%
104 | Sunrise | 10%
6 | Sweep | 10%
36 | Sweep Random | 10%
175 | Swirl | 10%
116 | TV Simulator | 10%
173 | Tartan | 10%
44 | Tetrix | 10%
13 | Theater | 10%
14 | Theater Rainbow | 10%
35 | Traffic Light | 10%
56 | Tri Fade | 10%
55 | Tri Wipe | 10%
17 | Twinkle | 10%
81 | Twinklecat | 10%
80 | Twinklefox | 10%
106 | Twinkleup | 10%
50 | Two Dots | 10%
113 | Washing Machine | 10%
140 | Waterfall | 10%
165 | Waverly | 10%
184 | Wavesins | 10%
127 | Waving Cell | 10%
3 | Wipe | 10%
4 | Wipe Random | 10%

508loopdetect, 2023-12-03:
[H0nzik1](https://github.com/H0nzik1)  thank you, i full agree to this list.

508loopdetect, 2023-12-24:
i finished my cabling and updated all units in the house to 14.0.  it is still annoying how fast are all animations are even at 0 or 1 effect speed.
i have 3 rows with 90 pixels each. strip density is at 60/meter.  
i can use some animations only at lowest setting, which is very sad. i want to try others, but the change is so fast it looks cheaper than stripes from home depot. 

i dont know why so few people complain about that. i think they use some animations some time and go for solid colour and let them change with time presets. to avoid problem with the wife....

softhack007, 2023-12-24:
Some effects (but not all) will become slower if you change the "target FPS" in LED settings.

blazoncek, 2023-12-24:
@softhack007 that is why I do not take this report seriously and would encourage you to dismiss it too.

508loopdetect, 2023-12-24:
@softhack007 
i understand the points you made, i cant confirm the complete list. i also dont blame it to the version 14 alone, i noticed it in earlier versions before, but did not wrote down the animation name. 
i will check the "target refresh rate (FPS)" in LED settings, because i dont use matrix yet. 

i´m sure with a generic slow down, we can satisfy all needs one and forever.  we need this anyway, because the next generations of matrix leds is on the way and they will be 4x faster. 

@blazoncek  
i dont like how you dealt  with this already before you last post, you can do what you want, but for me thats unacceptable. 
"Hi, unfortunately that was the intended change."   From WHO?  i dont think the users want a speed up at this level in some animations, or was there a poll, or something? 
If you want to change something in code, fine, but the speed for the user should be still almost the same, or usable in a normal way. Please check, if there is 255, range from 1-5 is still way to fast. even at 0.. 

i will give a manually tested list in a few days, which i thin k should not be so fast at level 1-5 from 255.

frohen Weihnachten und guten rutsch

blazoncek, 2023-12-25:
@508loopdetect it was *my* decision for the change as I am the author of that. If you want to *be in the loop* I am the one driving WLED development recently so am entitled to some decisions.

If you do not intend to put some effort into describing exactly what and where the change needs to be and just generically say "10% of the speed across all effects" (or agree to that) I just cannot take you seriously. It is a waste of my time.

And after all, WLED is open source and if you disagree with our (or my) decisions you are free to fork and modify source to your needs. No one is going to stop you. And if you get a lot of supporters you can PR those changes back upstream if you wish or just keep supporting your version.

508loopdetect, 2023-12-25:
your mindset is interesting, the change was because you are the author? ok.  without any request or user poll? 
ok, and if i want back the slower versions I HAVE TO:
- tell you all results in animations because of the changes YOU MADE?. 
- get A LOT OF supporters, just to come to the point i was some time ago but still belongs to your decision?
- 
WTF? 

OK no problem. i can live with that kind of BS. 
Please have fun to play the boss here. i thought this is a software for users, i was happy with the old speed.

softhack007, 2023-12-25:
@508loopdetect, @blazoncek peace, guys.

Please consider that most of us are not native speakers of englisch. Trying to "read between the lines" is often ending in bad misunderstanding.

The truth is, we sometimes modify effects. In the case of Tetrix, the time management was changed from "framerate based" to "time based". Framerate based effects are generally hard to maintain (some even say "the real evil"), so better to touch some effects and make them fit for the future. And yes, it happens that we have to re-think "what is a good baseline speed" when going from framerate to time based.

We don't do any polls before such changes. And yes, the person who created an effect (or did major updates) does have the freedom to decide on artistic aspects. 

If someone comes to us saying "would it be possible to slow this effects down" then it's something we try to look into. Of course we have a long backlog, and some things are more important than reducing the speed of Tetrix. @508loopdetect  I hope this makes sense to you.

@blazoncek is devoting a lot of his personal time into working on the source code of WLED, and this is something we all respect him for. At the same time he is not an autocrat, as important design decisions are always are made by a (small) team of regular contributors. We don't have a boss. But certain areas of the source code are primarily under one developers responsibility. Like in my case, If you want something changed in audioreactive, you'd be smart and convince me.

I see no bullies, no bosses, no bullshitting. Not even shitstorm troopers. It's just a small team of professional engineers doing WLED in their free time, and our backlog is really full already.

Let's keep this in mind when we talk. Please.

508loopdetect, 2023-12-26:
Thank you Frank for explaining more what was behind.  i know there is no time left, if you are involved in some os projects.
even if you are just testing and cant help in code things. 

have to relax, maybe i´m to old for fast animations and stick to the lake animation. thank you

H0nzik1, 2023-12-26:
> > [H0nzik1](https://github.com/H0nzik1) thank you, i full agree to this list.
> 
> Sorry @H0nzik1 and @508loopdetect, but the list you made does not help us developers. It seems you (blindly?) copied the complete list of effects from our KB, then wrote "10%" behind each.
> 
> In fact you even listed many effects that make me wonder if you ever tried them:
> 
>     * some did  not change at all since 0.13.x,
> 
>     * some in your list are not changing over time at all (like percent)
> 
>     * some are very slow already (like Aurora, Pacifica)
> 
>     * some are 2D (will not even run on single stripes)
> 
>     * some are audioreactive and not meant for "slow motion".
> 
> 
> In general, it might be that people have different expectations for the kind of animations that look "good". This could explain why we did not observe any "massive shitstorm" about speed and quality of WLED effects.
> 
> There is no way to satisfy everyone, but each of you is welcome to contribute to the project by making a PR, if you have a good idea on how to improve source code of some effects.
> 
> It might be possible (in 0.15. Development) to look into a generic "slow down" option, but this might take some time to develop.

You are funny. Do you think I have so much time to test every effect for you, just because you made it faster for some unknown reason? I have tested few I currently use, all are too fast even speed is set 1-10 out of 256!!!! 13.3 was better, but I was still using there speeds around 60 out of 256 at least. 

I am not your tester, I am just user and I am telling you, effects are too fast. If you dont listen users, and you are lazy to check speed of your effects by yourself okay, but I am staying at 13.3.

closed 2023-12-26 by blazoncek

softhack007, 2023-12-26:
> You are funny. Do you think I have so much time to test every effect for you

God news is, you are funny, too. This an open source community project, so your not testing anything for me, for Blaz or for Christian.

> I am not your tester

Well I am not your developer, either.

Looks like this topic is too heated emotionally, time to focus on something else.