wled/WLED #3558: Ethernet: wESP32 rev. (>= 7) not supported

skerner-dev, 2023-11-29:
### What happened?

Hello devs, hi community,

i noticed a bug or rather an **incompatibility** on one of the supported ethernet boards.

This concerns the (open source / HW) wESP32 POE board of Silicognition LLC. https://wesp32.com/

Not long after the initial release of the wESP32 the manufacturerer changed the PHY from LAN8720 to RTL8201 because of reliability issues.

The dev writes the following on https://wesp32.com/files/wESP32-Product-Brief.pdf (Page 6, Revision Changes):

> From revision 7 onward, the board uses an RTL8201FI Ethernet PHY chip for increased reliability
versus older revisions that used a LAN8720AI Ethernet PHY. A software change is necessary to
support the new PHY. In most cases, the only thing that changes is the definition of the PHY declared
in the code.

Since this change happened quite early after the release i guess of the wESP32 board sout there are with the RTL8201 PHY.

### To Reproduce Bug

Flash and configure WLED to a wESP32 rev 7 or greater.

### Expected Behavior

Connection via ethernet.

### Install Method

Binary from WLED.me

### What version of WLED?

0.14.0

### Which microcontroller/board are you seeing the problem on?

ESP32

### Anything else?

  quote from network.cpp
  
  ```
   // WESP32
  {
    0,			              // eth_address,
    -1,			              // eth_power,
    16,			              // eth_mdc,
    17,			              // eth_mdio,
    ETH_PHY_LAN8720,      // eth_type,
    ETH_CLOCK_GPIO0_IN	  // eth_clk_mode
  },
  ```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

softhack007, 2023-11-29:
Seems to be a duplicate of
* https://github.com/Aircoookie/WLED/pull/3444
* https://github.com/Aircoookie/WLED/issues/3057
*

skerner-dev, 2023-11-29:
Thanks for the info. I was not aware of these, sorry.

Would be very happy to see revs 7 and higher supported when WLED moves to IDF V4.

Best wishes :)

closed 2023-11-29 by skerner-dev