wled/WLED #3559: http-command "&A=0" is ignored, when "PL=xyz" is included

Clubsport, 2023-11-30:
### What happened?

Until 0.13.3 (verified), a http-command like http://IP/win&A=0&PL=1 was switching the main brightness to zero, which was perfect for usage with Loxone Dimmer outputs, as you can vary the dimmer value from 0 to 255, and send the value to WLED by http://IP/win&A=<v>&PL=1 .

In 0.14.0, the value of "A=0" is ignored, which means in this combination, you are not able to switch the lights off anymore.

### To Reproduce Bug

ESP-01 or ESP8266, V0.13.3 vs. V0.14.0 (including b6), use http- commands like written above.

### Expected Behavior

&A=0 should always set master brightness to zero, no matter what the rest of command line is.

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.14.0, WLED 0.14.0-b6

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2023-11-30:
The change comes withe the different preset loading mechanism.
Unfortunately we will not change that.

As a workaround please remove brightness (and on state) from your presets.

Clubsport, 2023-11-30:
That did the trick, THANK YOU! 👍

closed 2023-11-30 by Clubsport