wled/WLED #3563: Only 3 segments ON status saved

muttleydosomething, 2023-12-02:
### What happened?

I have a bookcase with 7 shelves. I illuminate all 7 segments I’ve created and the save the status overwriting data with current status. However when alexa turns on the lights only the first 3 segments come on.
I’m using Home Assistant and all 7 segments have an on/off button and when I turn them on manually everything works fine, I don’t think it’s me but i’m not ruling that out but I honestly suspect it’s worth submitting a bug report.

### To Reproduce Bug

Create more than 3 segments, turn them all on, save current status

### Expected Behavior

All segments should come on but only the first 3 seem to light up

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 14.0

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

ihavenonick, 2024-01-01:
I think I have identified the problem that is causing this behavior.
If a PRESET is switched off with Alexa, the status does not change according to the -number of the preset, but remains at the current number without -.

Example:
the preset number is 2 and if I switch this preset off manually via the WebIF, the state of the number changes to -2 and can therefore be switched on again with 2.
If I switch it off via Alexa, the status remains at 2 and setting it to 2 again causes no reaction.

Tested with ESP8266 and ESP32 with WLED 0.14.1-b2

softhack007, 2024-01-01:
> the preset number is 2 and if I switch this preset off manually via the WebIF, the state of the number changes to -2 and can therefore be switched on again with 2.

Where did you see this number? Presets in wled do not have negative id's. Actually a preset can be activated, but not de-activated; to de-activate a preset you would switch to another preset.

Please add some more details / screenshot to help us devs follow what you say.

ihavenonick, 2024-01-01:
I have integrated the WLED control into my smart home central IOBroker and can see all the statuses there
[image]

blazoncek, 2024-01-01:
"ps" with value of -1 means *no preset selected*
this happens when any of the values change (brightness, effect, effect parameter, color, etc.)

ihavenonick, 2024-01-01:
Is it possible to set the "-" flag even when Alexa switches it off?

blazoncek, 2024-01-01:
it is a value of negative 1 (-1), not a flag.
what it signifies is that no preset is currently selected and nothing more (all of the segments, effect parameters and colors remained unchanged since last update)

ihavenonick, 2024-01-01:
Yes, but why is it a difference if you switch with Alexa? If the status does not change to "-", I cannot switch on by resetting the preset number in IOBroker and that is the problem. I would also have to switch the data point on/off here. This is possible but more complicated.

muttleydosomething, 2024-01-01:
Sorry to interrupt but I’ve found I can ask Alexa to turn off and on the bookcase and all seven shelves do light up. The problem seems to be either; #1 if I switch the bookcase on as part of another group of lights in the bedroom. #2 It could also possibly be after a length of time (though i’m no too sure it’s this)

From my suspicions so far the problem seems to be as I mentioned in #1 Also I would like to add that there may be absolutely nothing wrong with WLED code whatsoever and it was only under suspicion. The issue could lie with the Alexa app. Still trying to narrow down the exact issue.

To aircookie, if you get to read this, I’d just like to say thank’s so much for your work it’s really appreciated.

ihavenonick, 2024-01-02:
I can also switch on and off with Alexa and have the same behavior when I want to switch devices in a different group, so that some segments no longer come on.
I control most with my smart home system via preset and I just noticed that Alexa switches this data point differently than controlling it via the WLED WebIF. This could also be the reason why it doesn't behave as expected.

muttleydosomething, 2024-01-21:
As the OP, I’d like to declare I still do not believe this issue completely resolved. I’ve been trying to break down what I believe is causing the problem. So far (and I could be mistaken) That after a certain length of time, or a specific time, maybe midnight. Turning all the segments on still fails.

Turning on the bookcase (all seven shelves) as part of a group of lights is now working BUT. If I turn the group of lights on the next day… I still only get the first two shelves of bookcase lighting up. I am using the current latest firmware as of this post as well as HA latest ver.

closed 2024-05-28 by github-actions[bot]