wled/WLED #3564: Toggling Power discards configuration

glenne, 2023-12-02:
### What happened?

When using the Power UI Button, it wipes the active led presentation.

### To Reproduce Bug

1. Create a custom preset such as `{"seg":{"i":[0,8,"FF0000", 10,18,"0000FF"]}}`.
2. Select the Solid Effect
3. Select the custom preset
4. Toggle Power Off
5. Toggle Power On
6. The Solid Effect is shown instead of the preset

### Expected Behavior

Use of the power UI Button should not change the led configuration.

### Install Method

Binary from WLED.me

### What version of WLED?

0.14.0

### Which microcontroller/board are you seeing the problem on?

ESP32-S2

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2023-12-02:
Using "i" segment API is only temporary. The outcome is not stored anywhere and as such is lost when WLED is turned off.
This is as designed.

glenne, 2023-12-03:
Ahh, unexpected but understood.  I'm using the i segment api to set a
region of leds at the peak of my roof to a different color.  I'm also using
Hubitat to turn the leds off and on morning and night.  Is there a
recommended way to create this type of customization so it will be restored
when power is toggled back on by the Hubitat controller? (except for my
playlist workaround)

glenne, 2023-12-06:
Thanks.  I was thinking the Segments listed on those screens were reconfiguring the LED Hardware segments.  I'll use that approach.

closed 2023-12-06 by glenne