wled/WLED #3565: Not yet reproduced WLED sounds issue

caladin, 2023-12-03:
### What happened?

Have used wled on esp32s3 wroom several times. 38 pin narrow version nodelabs, this was the first time I did it via USB c instead of usb micro.

Was trying the sound relative venison for the first time. Flashed from browser, wired it all up, everything worked, except the microphone. As soon as I picked a sound reactive effect, the leds just went blank. Acted exactly like I forgot to set the sound pin, or the mic was bad.

Definite for sure, I set the pin in the sound settings to match the pin I had wired it up to. But it acted exactly as if I had not.

Eventually noticed the note about power cycling after a sound pin change,, tried that , still no joy.

Spent hours trying to debug it, I tried everything including a different pin, different mic etc, hard power resetting the device multiple times, etc. Ran squelch down to to 1 and 0, and gain to 255. 

Used a 12 test light to verify that the mic was working by measuring the out put at the pin (Both 35 and 34), and saw the voltage change at the LED in the test light.

So I was 100% sure it had to be a setting/software issue, hardware was working. 

Eventually, I watched the video on you tube again, and followed along -exactly-, set the pin value again, but this time via the android app, and power cycled it (Again), and it worked perfectly.

I'd changed the value of the pin before, and I'd saved and power cycled it before. For some reason, it would only work from the android version of the app, not the web browser.

Going to flash a new esp32s wroom (38 pin), and try it again with my original mic, and I'll see if I can reproduce the issue while I'm at it. If I succeed, I'll come back and add that to the bug report.

### To Reproduce Bug

will attempt later today

### Expected Behavior

sound pin change takes affect after save and power cycle. (even better without a power cycle, but that sounds like a preexisting bug that got punted on)

### Install Method

Binary from WLED.me

### What version of WLED?

sound reactive 0.13.4

### Which microcontroller/board are you seeing the problem on?

ESP32-S3

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

softhack007, 2023-12-03:
btw, did you tick the "Enabled" checkbox in usermod settings for audioreactive ?

caladin, 2023-12-04:
yes, I may be off on that, I'm not a hardware guy, I'm a software guy.

This is the esp32 I'm running it on. I think it described it above, but I may not have.
https://www.amazon.com/gp/product/B0C8DBN29X/ref=ppx_yo_dt_b_asin_title_o04_s01?ie=UTF8&th=1

I figured out how to reproduce and "fix" the issue, Not sure if it's hardware or software.

had a working sound setup. 

Moved ground for Mike from power injection ground on strip to the ground pin next to p23. this fails (not sure why, but I wanted to test it)

Then when I moved it back, it was in full fail mode. Change the pin in ui , power reset back, fail. nothing I can do gets it working on pin 35 by altering settings and power cycling. android or web.

I had to change the pin to 34 in hardware and software(android), resetting power in between, before it started to work again.

after 34 worked, 35 did not. nothing i could do to make 35 work.

I for lack of a better term "locked" the sound up again, and was only able to fix it by unplug, let sit, restart, change pin 34->35, hard reset.

Then it worked again on pin 35.

feel free to do what you want with that info, I just felt I should report it.

softhack007, 2023-12-04:
> This is the esp32 I'm running it on.

That's a "classic esp32", so its not esp32-S3. I'll update your report accordingly.

On some boards, GPIO35 has some additional circuitry - so its possible that 35 would not work as analog input in your case.
Actually i've never seen problems with the browser UI for changing pin settings, so it might be something specific to your Computer setup.

Would you be so kind to tell us which build number you try to use? You'll find it on WLED info page - something like "Build 2310013"

softhack007, 2023-12-04:
Hi, 
a few hints:
* You can only use "ADC1" pins for analog microphones, see https://github.com/atuline/WLED/wiki/Sound-Settings#analog-input-pins; gpio2 will not work.
* some users have recommended to use alternative installers that make analog audio more reliable:
* * https://wled-install.github.io
* * https://kno.wled.ge/basics/install-wled-flasher

If possible, use an I2S  digital microphone like the INMP441. They work much better than any analog devices. https://github.com/atuline/WLED/wiki/Digital-Microphone-Hookup

closed 2024-04-21 by github-actions[bot]