wled/WLED #3567: Self organizing LED

BornTKill, 2023-12-04:
Dear,

i love WLED and was wondering if https://github.com/atomic14/self-organising-leds

Could be port to WLED ?

softhack007, 2023-12-05:
Hi, interesting tool :-)

While I don't see how this tool could be integrated/ported into WLED, it might be possible to to extend it, so it writes out a [ledmap file](https://mm.kno.wled.ge/advanced/mapping/) that WLED can use. Maybe contact the tool author, and ask him/her if that's possible to implement.

softhack007, 2023-12-05:
> https://github.com/atomic14/self-organising-leds

@ewoudwijma maybe this tool could be an interesting way to generate ledmaps, what do you think?

ewoudwijma, 2023-12-05:
That's for sure a good idea! Insane the repo has not been worked on for 2 years, but we can ask the author if he wants to work together on that .
Did you flashed it to a cam board and did it work?
I have one lying here. If you get it working I will also try

BornTKill, 2023-12-05:
Dear,

i flash on esp32 but not working. First Time i use platformio so maybe it’s my fault. Will be great if someone else can try.

BornTKill, 2023-12-05:
https://github.com/atomic14/self-organising-leds/issues/31

Valderramaz, 2023-12-05:
Hi Fellows. I tried this project before, but there are some improvements that had to be done before thinking to port this to WLED, unfortunatly I dont know how to program as per se, but love to tinker with code previously done. 

This year a found an alternative for my xmas tree, using an a ESP8266, a russian project called GyverTwink, much more mature, and performs very well, and much more easier to put it working.

If someone who speaks Russian could contact Alex Gyver regarding this, would be awesome.

cgreening, 2023-12-06:
I'm not sure that this would work with the ledmap system - this seems to be limited to just changing the order of the LEDs.

What you really need to do is to work out in 2D (and possibly 3D) space where the LEDs are and then "project" an image onto the LEDs.

There are a lot of projects out there (mine is just one of many) that attempt to do this.

The basic idea is pretty simple, you point a camera at the LEDs and then flash them so you can identify the location. There are a lot of ways to do this - I took a very naive simple way and just flashed each LED one at a time and looked at the difference between an unlit and lit image.

Other people flash a code to identify each LEDs or use a sequence of colors. There are even more sophisticated methods that will let you identify multiple LEDs at once which is much faster when you have a lot of LEDs.

For 3D mapping you would need multiple cameras and it all gets a lot more difficult - which is why all the projects I've found tend to stick to 2D mapping.

I don't really have the bandwidth to work on anything, and if WLEDs doesn't support the mapping of an image onto arbitrarily positioned LEDs then it's a bit of a non-starter.

Kamal-Sonani, 2025-09-19:
use mobile's camera & remove hardware dependency , send next leds data from web page to controller using websocket (artnet if raw android app)

web socket need some additional code update in wled source tree , raw android app can be achievable method without touching source of WLED 
send next artnet led , capture , continue....
after process done need to manage that mapping in wled