wled/WLED #3568: Button press or analog change should be ignored when OFF

msn444, 2023-12-04:
### What happened?

When the WLED instance is in the OFF state, either due to UI or GPIO switch: pushbutton presses and even Analog input 
 (brightness) changes will cause WLED to enter the ON state.

### To Reproduce Bug

Set up an analog input to change global brightness. Turn WLED OFF using the UI or GPIO switch. Turn the potentiometer -- the WLED state will switch to ON.

### Expected Behavior

If the instance is OFF, I think it should stay OFF until explicitly turned on. Otherwise an errant bump of the potentiometer or pushbutton will cause the lights to come on when not desired.

### Install Method

Binary from WLED.me

### What version of WLED?

0.14.0

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2023-12-05:
That is by design and will not change.

msn444, 2023-12-05:
What led to that design decision? Can't there be an OFF state that really means OFF? I think WLED is a great product and is full of great design decisions, but I don't think this is one of them.

blazoncek, 2023-12-05:
Having a desk lamp with only a potentiometer.

msn444, 2023-12-05:
Sure, but that use case isn't precluded at all by providing a conventional ON/OFF state.

My exact application is a desk lamp with a potentiometer AND an on/off switch. I don't want the pot to override the switch.

Perhaps this is more of a feature request than a bug.