wled/WLED #3570: Improve Multi-strip support (option to use FastLED's I2S 24-way parallel driver)

rorosaurus, 2023-12-05:
**>Is your feature request related to a problem? Please describe.**
Existing multi-strip support is limited, especially for ESP32-S3, ESP32-S2, and ESP32-C3. Though existing support is more flexible (works for clocked chips, or different protocols), only having 2/3/4 strips per ESP32 can be restrictive for many projects - often leaving significant performance on the table. This could increase complexity (extra unnecessary ESPs) and construction time (awkward data return lines) for certain projects.

**>Describe the solution you'd like**
It would be nice if we could create an option to use [/u/Yves-Bazin](https://www.reddit.com/u/Yves-Bazin) and [/u/samguyer](https://www.reddit.com/user/samguyer)'s [24-way parallel driver for ESP32](https://www.reddit.com/r/FastLED/comments/bjq0sm/new_24way_parallel_driver_for_esp32/) to drive RGB only, clockless strips of the same chip type. This would optionally increase the max strips to 24 for ESP32, ESP32-C3, and ([I think? I haven't tested](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-reference/peripherals/i2s.html)) ESP32-S2/S3. Then we could enjoy the benefits of both worlds! I suspect many multi-strip installs meet those requirements, so such a feature would offer those projects extra density/resolution/flexibility.

**>Describe alternatives you've considered**
Sometimes it's not possible to join strips together in a serpentine pattern (or 2D matrix) for aesthetic or functional reasons. For example, I'm building a [beaded door curtain](https://www.google.com/search?q=beaded+door+curtain&tbm=isch) where chaining together the data lines at the bottom of each strip would result in loops that can snag upon entry. I've worked on other projects in the past that used 18 strips, where the physical layout of the strips would have made it very difficult to chain together data lines. That forced me to fallback to using FastLED when I would've truly loved to enjoy all the benefits of WLED adds to an art install!

**>Additional context**
#104 addressed the initial implementation of multi-strip support
[Current Multi-strip Support - WLED](https://kno.wled.ge/features/multi-strip/)
[clockless_i2s_esp32.h](https://github.com/samguyer/FastLED/blob/08388047ac3e0ebcaafe71563ae2a55421d70c02/src/platforms/esp/32/clockless_i2s_esp32.h)

Thanks everyone for making WLED the best! It's been a privilege to watch this project grow and develop over the years! Here's to many more!

blazoncek, 2023-12-05:
WLED does not use FastLED drivers.

rorosaurus, 2023-12-05:
Yes, I'm proposing a new feature to use FastLED drivers when it's beneficial.

blazoncek, 2023-12-05:
Changing the LED driver would mean almost rewrite WLED from scratch. Not in any foreseeable future.

There is, however, a similar thing in recent release of NeoPixelBus. Implementing that may take a while.

softhack007, 2023-12-05:
> > Yes, I'm proposing a new feature to use FastLED drivers when it's beneficial.

We cannot "jump" between drivers; WLED uses Neopixelbus for addressable LEDs. If you have an example piece of code for a new driver, the maintainer of [NeoPixelBus](https://github.com/Makuna/NeoPixelBus) (makuna) might be happy to try and implement something similar.

> There is, however, a similar thing in recent release of NeoPixelBus. Implementing that may take a while.

I remember there is a "8x/16x parallel" driver in NPB. https://github.com/Makuna/NeoPixelBus/wiki/ESP32-NeoMethods#parallel-channels

<br/>

I've made some experiments with I2S parallel driving mode, half a year ago. There are a few drawbacks for general use in WLED:

1) RAM needed is simply insane. I've tried 1200pixels, evenly distributed so all "8x parallel" channels were in use. Free RAM in wled dropped below 40Kb, making our software absolutely unstable so it crashed every few minutes (WiFi dies, UDP out of memory, webserver crashes, etc).
2)  "8x/16x parallel" driving is only possible right now on "classic esp32", because the driver abuses a specific mode of the I2S units. According to Makuna, this specific I2S mode is not available on -S3/-S2/-C3, so won't help with new MCUs.

<br/>

Some references:

* https://github.com/Makuna/NeoPixelBus/wiki/ESP32-NeoMethods#parallel-channels

* https://github.com/Makuna/NeoPixelBus/issues/648
* https://github.com/Makuna/NeoPixelBus/issues/598
* https://github.com/Makuna/NeoPixelBus/issues/605
* https://github.com/Makuna/NeoPixelBus/issues/697#issuecomment-1548126832

blazoncek, 2024-03-29:
Played with parallel I2S output and as of NPB 2.7.9 it is useless for anything beyond 250 pixels per output (for 8 channel I2S, 16 channel will have limit of 125 pixels).
Reason being huge RAM demands from I2S DMA buffers.

rorosaurus, 2024-04-17:
Useless as in poor refresh rate? Thank you for that investigation!

In SmartMatrix and others, I'm familiar with using external PSRAM to help alleviate the limited DMA memory. Is there any chance the bottleneck you're seeing is of a similar type that could be addressed in this way? I know quite a few dev boards that come with PSRAM, which could be fair requirement for mass parallel output.

blazoncek, 2024-04-17:
Please provide links.

softhack007, 2024-04-17:
> n SmartMatrix and others, I'm familiar with using external PSRAM to help alleviate the limited DMA memory. Is there any chance the bottleneck you're seeing is of a similar type that could be addressed in this way?

I don't think that PSRAM will help. We had a version that could use PSRAM for LED buffers and effect runtime data. However, on an esp32 WROVER board (rev.3) this resulted in heavy framerate drops --> Like 60fps when using normal RAM, but down to 5-15 fps when using PSRAM. 

I haven't checked if the performance on -S3/-S2 is better, but the bad results on classic esp32 made us conclude that PSRAM is not good for fast changing data. In WLED we found another use for PSRAM - JSON data and "ajax" style UI communication.

blazoncek, 2024-04-17:
BTW I'm working with @Makuna on resolving this issue. I have WLED working with 8x 400 pixels ATM.

coliniuliano, 2024-05-08:
@blazoncek Any update? Do you have a branch you're working on that I can play with or take a look at?
I'm eagerly awaiting more strips on the S3 and happy to contribute.

blazoncek, 2024-05-08:
The template is available in 0_15 and you'll need to uncomment x8 I2S and replace existing I2S. It should be obvious.

Makuna, 2024-05-08:
Note, C3 and S3 are not supported with parallel yet.  They changed the API due to pulling the parallel support out of the i2s hardware.  It's there in another piece of hardware, so requires more research/code to support.

This is being tracked by https://github.com/Makuna/NeoPixelBus/issues/605

blazoncek, 2024-06-19:
Available in 0.15.0-b3 and later, build 2406120 or greater.

closed 2024-06-23 by blazoncek