wled/WLED #3576: mqtt status takes a long time to update

viktor6, 2023-12-08:
### What happened?

mqtt status takes a long time to update
esp8266   wled/wled/status   30 seconds
esp32      wled/wled/status   2 minutes  which is very long

### To Reproduce Bug

on/off esp32 wemos D1 

### Expected Behavior

30 c

### Install Method

Binary from WLED.me

### What version of WLED?

Installed version: WLED 0.14.0 (ESP32-D0WDQ6 build 2310130)

### Which microcontroller/board are you seeing the problem on?

ESP32

### Anything else?

On esp8266 it makes a status update from offline to online in 30 seconds
On esp32  it makes a status update from offline to online in 2 minutes

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

viktor6, 2023-12-08:
but on my esp32 it updates consistently in 2 minutes. I tried 3 different esp32s and more than once

viktor6, 2023-12-08:
I need to track when wled appears on the network and for further automation

dosipod, 2023-12-08:
Okay i so what you mean  when you listed on/off  you actually mean remove power or make it offline from the network  
[image]
I can see use case for car garage

viktor6, 2023-12-08:
And on esp8266 it is about 30 seconds

blazoncek, 2023-12-08:
You don't seem to understand MQTT. `/status` topic is **retained** with LWT set.
When WLED device is without power you will receive LWT (which is "offline"), when it is powered on you will receive "online". WLED will not re-broadcast it on its own but only together with any other state change (see example below).

[image]

dosipod, 2023-12-08:
> And on esp8266 it is about 30 seconds

I also got the same strange result you pointed out when I tested with esp8266 , it will be reconnected after i remove power  in very much less time  then esp32 . Must be something with mqtt re-connection on esp32  could be heavy after power down  , I did not even trap for offline /online but only for wled/WLED/g as you see in the example which is the same test I did on esp32  

[image]

dosipod, 2023-12-09:
@viktor6  
I found something you may want to try , once I changed the   MQTT_KEEP_ALIVE_TIME value in  wled00/mqtt.cpp the status change on esp32 is much faster now  , Bin and change is here 
https://github.com/dosipod/wled-dos-mod/actions/runs/7151014493
[image]

closed 2024-07-23 by github-actions[bot]