wled/WLED #3577: UCS2904 RGBW pucks not working properly in v14+

elylv, 2023-12-08:
### What happened?

I have these puck LEDs:
https://www.aliexpress.us/item/3256805073853110.html

On v13.1 I can set them to TM1814 (GRB) and control them well.
On v14 (including betas), they will just show orange with some of the lights flickering.
_Sometimes_ if I mess around swapping B&W, G&W or R&W a few times, they will start responding (although the "swap" setting does not seem to persist in the interface)... but eventually, after the lights being off for a while, they will turn back on orange.
Sometimes, when attempting to change to a solid color, the lights will change color but immediately revert to orange.

I have seen people say using SK6812 on these lights works, but have been unable to reproduce (the lights just cycle through all colors, regardless of setting).

### To Reproduce Bug

Connect lights to WLED (via DigQuad) on version 14 or later.

### Expected Behavior

Lights to behave same way they do under 13.1

### Install Method

Binary from WLED.me

### What version of WLED?

14.0 -> b5

### Which microcontroller/board are you seeing the problem on?

ESP32

### Anything else?

I am colorblind, so my description of the light being orange may be off slightly.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

softhack007, 2023-12-09:
> Interested in how it's working in v13

Our LED driver is NeoPixelBus. In v0.13.x we used an older release, which by coincidence maybe worked better with your lights. However from WLED side we don't work on signal protocol level. 

It might help if you first make a feature request for [NeoPixelBus](https://github.com/Makuna/NeoPixelBus) to support your chip, and later we might be able to integrate the new driver.

elylv, 2023-12-09:
Turns out the chips in the lights linked above are TM1814 rather than UCS2904, hence the success with TM1814 protocol, however it was very intermittent.  That is until I disabled Global LED Buffer on the suggestion of someone else with those lights, and now it is working correctly.

I'm not sure if this is a bug with Global LED Buffer in v14, or if the lights don't support that somehow.  Regardless, it seems to be working now and the lights I bought were somewhat mislabelled by the seller.

blazoncek, 2023-12-09:
TM1814 will be fixed in 0.14.1. The code is already in place.
As a workaround use effect other than solid.

closed 2023-12-09 by elylv