wled/WLED #3578: Allow white channel control while effects are applied

gaaat98, 2023-12-09:
**Is your feature request related to a problem? Please describe.**
Depending on the auto-white calculation mode that is selected the white channel of RGBW strips (tested on RGBW PWM, WLED v0.14.0-b1) is different and cannot be manually controlled. 

- When the mode is set to None the white channel does not turn on and changing the value of the white channel slider does not turn it on.
- When the mode is set to Dual the white channel is auto calculated no matter the position of the slider (so not only when the slider is set to 0) and the intensity of the white is not affected by the slider

The effect tested on the RGBW PWM strip was Colorloop with different palettes, but I also have verified on a SK6812 (WLED v0.14.0) strip that also led-addressable effects lack the control of the white channel intensity, in this case it appears to me that the white channel is always autocalculated independently of the mode selected.

**Describe the solution you'd like**
Depending on the feasibility some solutions could be:
- Use the slider value as a maximum value for white channel, so depending on the effect the white channel will be interpolated from 0 to slider_value
- Use the slider value as a fixed value for the white channel, so if the led is ON it uses that white value, if OFF the white channel is also off

**Additional context**
A side note: imho the auto white calculation should not be activated by the value of the slider: there are cases where a user could want to temporarily disable the white channel despite having the auto white calculation enabled in other, hence setting the slider to 0. 
With the current setup this is not possible. A better approach would be to have something like a checkbox labelled as "Auto" near the slider. Checking it disables the slider and enables auto-white calculation, when unchecked the slider is active and can be set to zero.

I understand that these suggestions may be unfeasable to implement so no worries! In any case thank you for your time and thank you a lot for this awesome project!

blazoncek, 2023-12-09:
There are two "auto white" options in LED settings.
One is per-output:
[image]
the other is global override:
[image]

Setting first to "None" and the second to "Disabled" will allow manual white channel control.

gaaat98, 2023-12-09:
Thank you for the answer! Yes, the settings you suggest indeed should allow for manual white channel control, however this manual control appears to be working only when the default color palette is specified.

For example, if I select the effect "Colorloop" and I apply the palette "Aurora 2" the white channel slider has no effect, neither when the mode is set to Dual nor None (the global override has been always set to disabled in my case).

Is this an intended behaviour?

blazoncek, 2023-12-09:
I have tested and indeed when you select Colorloop (or any other effect) white channel control is lost.
That may be unfortunate consequence of the fact that most (if not all) effects do not handle white channel separately and will set it to 0 (black) regardless of the white channel value in UI.

The only effect that correctly handles white channel is Solid.

This issue may be difficult to solve as it may require rewriting of all effects (or functions that deal with color selection) to separately handle white channel.

softhack007, 2023-12-09:
Actually there is another option that will - in most effects - kill the white channel of pixels: "Use global LED buffer". The buffer is usually recommended for better color quality, but it only uses a 24bit (3color channels) buffer. So in effects that use getPucelColor() to get the last color of pixel, this operation will return white=0.

blazoncek, 2023-12-09:
> Actually there is another option that will - in most effects - kill the white channel of pixels: "Use global LED buffer". The buffer is usually recommended for better color quality, but it only uses a 24bit (3 color channels) buffer. So in effects that use `getPixelColor()` to get the last color of a pixel, this operation will return white=0.

Global buffer is 32-bit in case of RGBW and 24-bit in case of RGB only. It has no effect on the issue above.
The real problem are support functions that only handle RGB and strip W channel. Most effects also do not deal with W channel specifically.

softhack007, 2023-12-09:
> Global buffer is 32-bit in case of RGBW and 24-bit in case of RGB only. It has no effect on the issue above.

Ah, it seems I still had the old implementation in mind, that used an array of CRGB for buffering. Thanks for putting it right.

blazoncek, 2023-12-09:
BTW I think I've solved the problem with support functions.
Not with effects though.

gaaat98, 2023-12-19:
@blazoncek thank you a lot!

> A side note: imho the auto white calculation should not be activated by the value of the slider: there are cases where a user could want to temporarily disable the white channel despite having the [...]

Do you think that this way to control the white channel in the UI could be adopted in the future?

blazoncek, 2023-12-19:
Use Dual mode. If white is at 0 then Auto is used, otherwise user input is used.

BTW, @gaaat98 you are welcome to buy me a beer. 😉 Link on the main page.

blazoncek, 2024-03-22:
Available in 0.15.0 beta 1.

closed 2024-03-22 by blazoncek