wled/WLED #3581: blink random color when turning on with relay

doronazl, 2023-12-12:
### What happened?

im using a SSR to control my PSU and turn it off when not in use, it works fine but im getting a blink sometimes when turning the led on, been reading about it and found this thread
https://github.com/Aircoookie/WLED/issues/812
seems this was solved there but trying to edit the changes proposed i find things are different in the current code so im stuck

how do i get this done , id like to add a 200ms delay 

### To Reproduce Bug

turning on with a relay

### Expected Behavior

described

### Install Method

Binary from WLED.me

### What version of WLED?

0.14

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

doronazl, 2023-12-12:
ive also tried the multirelay usermod, having problems with it, i set the gpio relay and hit save, it saves but after a refresh the gpio pin return to unused, every numeric one act this way, the only ones that actually save are P0-P7, which i dont think available for my board, esp32wroom

softhack007, 2023-12-12:
@doronazl could it be you are regularly deleting / changing your questions? I got some questions by e-mail but they are missing here, so hard to help you.

About multi-relay:
* the delay time is an integer (whole number) for seconds. So you can have 0sec, 1sec, 42sec.  But you cannot use fractions like 0.34567 sec.  If you want finer control, either modify the source code, or make a feature request.
* (deleted question) if you enter GPIO pins in the usermod, make sure to not use the same pin anywhere else. So in LED settings, you have to configure the relay pin as unused (-1).
* (deleted question 2) P0...P7 are for the port expander, if you use one. Otherwise don't use P0 ... P7.

doronazl, 2023-12-12:
> @doronazl could it be you are regularly deleting / changing your questions? I got some questions by e-mail but they are missing here, so hard to help you.
> 
> About multi-relay:
> 
> * the delay time is an integer (whole number) for seconds. So you can have 0sec, 1sec, 42sec.  But you cannot use fractions like 0.34567 sec.  If you want finer control, either modify the source code, or make a feature request.
> * (deleted question 1) if you enter GPIO pins in the usermod, make sure to not use the same pin anywhere else. So in LED settings, you have to configure the relay pin as unused (-1).
> * (deleted question 2) P0...P7 are for the port expander, if you use one. Otherwise don't use P0 ... P7.

My apologies, some issue i got resolved so i just deleted it.
All i need now is a way to do ms, not seconds.
Wouldnt know where to modify and what 🫤

softhack007, 2023-12-12:
> All i need now is a way to do ms, not seconds.
> Wouldnt know where to modify and what 🫤

Before you modify code, it would be smart to first verify that the MultiRelay usermod does indeed solve your problem, by using 1sec or 2sec as the delay value.

If that question is answered "yes", you could try to modify this line in the source code:
https://github.com/Aircoookie/WLED/blob/1dab26bcbcac051f2b7be47a2d5c757a9938bf1f/usermods/multi_relay/usermod_multi_relay.h#L225

Changing the `* 1000` into  `* 100` modifies the resolution of the input value (from 1sec to 100ms units); so if you now enter "3" in the UI, it actually means 0.3 sec.

doronazl, 2023-12-12:
Will do!! Thank you so much

doronazl, 2023-12-13:
> > All i need now is a way to do ms, not seconds.
> > Wouldnt know where to modify and what 🫤
> 
> Before you modify code, it would be smart to first verify that the MultiRelay usermod does indeed solve your problem, by testing 1sec or 2sec as the delay value.
> 
> If that question is answered "yes", you could try to modify this line in the source code:
> 
> https://github.com/Aircoookie/WLED/blob/1dab26bcbcac051f2b7be47a2d5c757a9938bf1f/usermods/multi_relay/usermod_multi_relay.h#L225
> 
> Changing the `* 1000` into `* 100` modifies the resolution of the input value (from 1sec to 100ms units); so if you now enter "3" in the UI, it actually means 0.3 sec.

so basically if i change it to *1 then i can treat my input in there as ms right ? 
so inputting 300 would mean 300ms ?

softhack007, 2023-12-17:
Just curious - what is the magic about 0.2 seconds delay? Would the flicker still happen with 1 second delay, or longer?

doronazl, 2023-12-18:
> Just curious - what is the magic about exactly 0.2 seconds delay? Would the flicker still happen with 1 second delay, or longer?

the UI wouldnt let me go lower than 0.2, i read somewhere even 50ms would be fine to avoid the flickering

a 1 second would work as well but its too long duration and its noticable.

KlausHans, 2024-01-15:
It would be very nice to add a delay as discussed here https://github.com/Aircoookie/WLED/issues/346 unfortunately it is closed even though the workaround proposed there is a bad one imho.
To simplify things, wouldn't it easily be possible to add a static, blanket delay of 100 or 200 ms if "Relay GPIO" is set in the config? That would be so nice and would prevent recurring issues like this.

closed 2024-05-23 by github-actions[bot]