wled/WLED #3583: Control using two ESP32 single RGB LED bulbs

Marcin685, 2023-12-12:
Hello. I have 18 Tuya RGB LED bulbs installed in the ceiling at home. I would like to control them using the UDP DDP protocol. I uploaded the ESPHOME DDP software to them from this link https://github.com/KaufHA/common/blob/main/components/README.md#ddp. Everything works when I want to control them from HomeAssistant, xLights, manually. Each bulb is assigned a person's IP address. I would like to control all 18 LEDs via WLED. I can use an ESP32 with 10 outputs for this, but I need 18 outputs. Why isn't this possible in WLED? Maybe it's possible to recompile e.g. to 20 in such cases? Maybe there is such a version? Each of my diodes is only 1 long, and for 18 LEDs it will be 18, so very little considering the possibilities of WLED. Of course, I can connect a second ESP32 controller, which will give me a maximum of 20 outputs. I want to make these 18 LEDs form one string, not separate 9+9. I don't know if WLED has such an option to create one LED string using two ESP32. I have tried various configurations, but I cannot set ESP#1 as primary and ESP#2 as slave. Thanks in advance for any tips.

blazoncek, 2023-12-12:
Too much internal work and will most likely not be implemented at any time.

ingDIY, 2024-01-09:
Hello,
maybe it's possible modifying WLED max busses:
`build_flags = ${env:esp32dev.build_flags} -D WLED_MAX_BUSSES=20 -D DEFAULT_LED_COUNT=1`

cheers

blazoncek, 2024-01-10:
Not beyond 10