wled/WLED #3584: ESP32-S3-N16R8 crash when install WLED

rcavezzale, 2023-12-12:
### What happened?

I'm tried install wled thought install.wled.me without success

1. unable to configure Wireless data
2. in console seems that the ESP crashed

Thanks for great project :)

### To Reproduce Bug

try to install WLED on the board

### Expected Behavior

that works :)

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.14.0

### Which microcontroller/board are you seeing the problem on?

ESP32-S3

### Relevant log/trace output

```shell
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xa (SPI_FAST_FLASH_BOOT)
Saved PC:0x403cd98e
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x43c
load:0x403c9700,len:0xbec
load:0x403cc700,len:0x2a3c
entry 0x403c98d8
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xa (SPI_FAST_FLASH_BOOT)
Saved PC:0x403cd98e
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x43c
load:0x403c9700,len:0xbec
load:0x403cc700,len:0x2a3c
entry 0x403c98d8
```

### Anything else?

I hope that for this board there is other procedure :)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2023-12-12:
ESP32-S3 is not yet fully supported and is in experimental phase.
Please use WLED [forum](https://wled.discourse.group) or Discord for help and support questions.
Closing.

closed 2023-12-12 by blazoncek

jasmohan-narula, 2024-04-25:
@HAlover99 Similar thing is happening with me. Let me know if you found any solution?

I tried flashing WLED_0.14.3_ESP32-S3_8MB.bin on ESP32-S3-N16R8

When the board crashes, I get this in serial console
`
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x400454d5
SPIWP:0xee
mode:QIO, clock div:1
load:0x3c100020,len:0x49040
ets_loader.c 78
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x10 (RTCWDT_RTC_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:QIO, clock div:1
load:0x3c100020,len:0x49040
ets_loader.c 78
`