wled/WLED #3588: DDP Sync also for white, CCT and RGB-CCT

traeu, 2023-12-14:
**Is your feature request related to a problem? Please describe.**

No problem. Just an idea for enhancement.

Current situation: I use a total of four H801 LED-Controller. They each have 5 PWM outputs.
I use them to control different PWM-LED-Spots (RGB-CCT, RGBW and CCT spot types). The first controller is connected to one RGB-CCT spot, the second one powers one RGBW spot, and the last two ones each control two CCT white spots.
Currently, by using the sync function on all controllers, I can change white (intensity, and temperature on CCT) of all spots simultaneously and also change RGB-colors syncronized.

It would be really cool if one of my controllers could be the "master" and be aware of the other (CCT)-LED spots. Then it would be possible to use the "master's" web-UI to run effects using all the spots or control each spot independently.

This currently works for RGBW-spots. I can setup two controllers with "PWM RGBW" config, then use one of them as "master" by adding the other controller as virtual second LED to it by using "DDP RGBW". Then one controller knows about two LEDs and can play effects on them. 
However, this does not work with CCT-LEDs. There is no "DDP CCT" or "DDP RGB+CCT" LED option for connecting virtual CCT-white or RGB-CCT spots.

**Describe the solution you'd like**
It would be great to have additional DDP options: "DDP CCT", "DDP RGB+CCT" and maybe also "DDP white".
Right now, it is possible to have two CCT-white spots physically connected to a WLED controller, but it's not possible to connect a virtual CCT-LED.

**Describe alternatives you've considered**
As an alternative for "DDP white" it is possible to use "DDP RGBW" instead, according to a quick experiment.
Also, "DDP RGBW" can be used to address a WLED controller that physically controls CCT-white spots. RGB is ignored and white is received. But, of course, then no color temperature values are transmitted, and the receiving controller converts receiving 100% white over DDP into 50% cold and 50% warm white PWM for the CCT-white spots.

blazoncek, 2023-12-14:
DDP protocol does not specify CCT.
http://www.3waylabs.com/ddp/

traeu, 2023-12-15:
Thanks for your answer! 
I changed the title of my request, I just thought that DDP would be a suitable solution because it is already used for virtual RGB and RGBW LEDs. What I would love to see is an option to add virtual (RGB+)CCT LEDs, this is the reason why I opened this request. Maybe DDP is not the best option for this or even no option at all.
I am no programmer and never used DDP before I tried it with WLED, so I don't have any knowledge about this protocol.
I read the protocol specification and it describes that the first bit of byte 2 can be set to 1 to use a custom type alternatively to the standard types RGB/HSL/RGBW/grayscale.
Maybe it would be possible to add a custom RGB+CCT type to WLEDs DDP-function? It would of course only work between WLED devices and not with other DDP-compatible lights.