wled/WLED #3589: Using JSON API right after UDP Realtime protocol doesn't work

kryptan, 2023-12-14:
### What happened?

If you call JSON API right after using UDP realtime protocol then led strip gets stuck on the color set by the last UDP packet. Neither does it return to normal mode after a timeout nor does the JSON API call works. Subsequent JSON API calls do work though.

### To Reproduce Bug

1. Send UDP realtime packet to WLED. e.g. DNRGB.
2. Wait 100 ms.
3. Call `json/state` with `"on": false`.

Example Rust code: https://github.com/kryptan/wled-bug/blob/master/src/main.rs

### Expected Behavior

JSON call should have an immediate effect or returning to normal mode after a timeout should work.

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.14.0 (build 2310130)

### Which microcontroller/board are you seeing the problem on?

ESP8266, ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2024-01-02:
Try adding
```c++
  } else {
    strip.trigger();
```
or
```c++
  } else {
    strip.show();
```
between lines 197 and 198 in `udp.cpp` from *main* branch, compile and report back.
After this line:
https://github.com/Aircoookie/WLED/blob/4f55be4f19daf6c81a167f5bf6bd20ff22afb664/wled00/udp.cpp#L197

blazoncek, 2024-01-04:
Please provide feedback.

blazoncek, 2024-03-04:
No feedback provided and most likely fixed.

closed 2024-03-04 by blazoncek