wled/WLED #3593: USERMOD Multirelay prevents sound reaction effect from turning on

doronazl, 2023-12-17:
### What happened?

my setup is WS2815 and esp32, using a SSR to cut power to the psu whenever the lights are off.
did some testing and manged to focus the issue onto the multirelay usermod

what happens is as follows - 
id be playing music with Gravimeter effect, lights turn on and work fine by the rythm, then id turn off wled, it turns off, cut power to the psu just fine, issue starts when i turn it back on, if i turn on wled and im on the SR effect, the light will not turn on at all, and looking at the SSR, the led wont turn on so its not engaged, hence no 3.3 signal coming from wled.
i then switch to solid color effect, the lights will turn back on fine, i then switch back to gravimeter, it works fine.

the problem is only happening when i try to turn on wled straight into the SR effect.
i then turned off usermod multirelay and switched the relay into the main setting under led prefrences, it works fine this way, cuts off power to the psu but just not through multirelay mod.

before you ask why im not just using the main setting for relay - i need the time delay option available on usermod

### To Reproduce Bug

explained above

### Expected Behavior

to work just like the main option

### Install Method

Self-Compiled

### What version of WLED?

0.14

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

softhack007, 2023-12-17:
Actually it could be the other way around - if mutirelay does not send the switching signal to turn the relay on, then it might be that multirelay "loop" function does not run s expected.

It would happen with 99% of all effects; "solid" is a special case because it only trigger a LEDs update 3 times per second.

Please try to modify this code
https://github.com/Aircoookie/WLED/blob/1dab26bcbcac051f2b7be47a2d5c757a9938bf1f/usermods/multi_relay/usermod_multi_relay.h#L516-L520

To look like this
```C++
 static unsigned long lastUpdate = 0;
 if (!enabled || (strip.isUpdating() && millis() - lastUpdate < 100)) return;

  if (millis() - lastUpdate < 100) return;  // update only 10 times/s
  lastUpdate = millis();

```

And report back if it works better.

The same problem exists with several usermods - if you have a long LED strip, the strip is always updating, and the usermod cannot run.

doronazl, 2023-12-17:
> Actually it could be the other way around - if mutirelay does not send the switching signal to turn the relay on, then it might be that multirelay "loop" function does not run as expected.
> 
> It would happen with 99% of all effects; "solid" is a special case because it only trigger a LEDs update 3 times per second.
> 
> Please try to modify this code
> 
> https://github.com/Aircoookie/WLED/blob/1dab26bcbcac051f2b7be47a2d5c757a9938bf1f/usermods/multi_relay/usermod_multi_relay.h#L516-L520
> 
> To look like this
> 
> ```c++
>  static unsigned long lastUpdate = 0;
>  if (!enabled || (strip.isUpdating() && millis() - lastUpdate < 100)) return;
> 
>   if (millis() - lastUpdate < 100) return;  // update only 10 times/s
>   lastUpdate = millis();
> ```
> 
> And report back if it works better.
> 
> The same problem exists with several usermods - if you have a long LED strip, the strip is always updating, and the usermod cannot run.

line 516 starts with 2 spaces whereas the code you sent me starts with 1, will it be ok ?  just replace straight up ?

doronazl, 2023-12-17:
> ```c++
>  static unsigned long lastUpdate = 0;
>  if (!enabled || (strip.isUpdating() && millis() - lastUpdate < 100)) return;
> ```

ok i did it, and it fixed the problem, turning on straight to the effect

doronazl, 2023-12-17:
Wonderful, working great now, thank you!
hope others run into this post or the changes will be added to the next releases

doronazl, 2023-12-17:
question, is there a way to make the turning on use the transition time ? 
seems that whatever i put in the transition effect doesnt change anything when turning it on after all power was cut.
if i turn it off and the immediately on again, it will dim down and up , but when turning it off, waiting a few seconds and then turning back on , it turns back on in a flash straight to the set brightness , no dimming effect

softhack007, 2023-12-17:
> ok i did it, and it fixed the problem, turning on straight to the effect

@blazoncek this is the "be nice but not too nice" thing we already added in several usermods. As multirelay is from you, it's your decision to update the code for the next release, or leave it as is. What's your preference?

doronazl, 2023-12-18:
checked and this happens both with 

> > ok i did it, and it fixed the problem, turning on straight to the effect
> 
> @blazoncek this is the "be nice but not too nice" thing we already added in several places. As multirelay is from you, it's your decision to update the code for the next release, or leave it as is. What's your preference?

do you know what i mean about the initial coming on effect with the relay ?

softhack007, 2023-12-18:
> do you know what i mean about the initial coming on effect with the relay ?

Actually no - normally I don't  see any transition (crossfade) at power-on. Slow crossfade only happens when changing effects. Maybe @blazoncek has an idea ...

doronazl, 2023-12-18:
> > do you know what i mean about the initial coming on effect with the relay ?
> 
> Actually no - normally I don't see any transition (crossfade) at power-on. Slow crossfade only happens when changing effects. Maybe @blazoncek has an idea ...

here
so this is turning wled off and on pretty quickly
https://streamable.com/3tge0i

and this is waiting a few of seconds in between commands
https://streamable.com/j12k46

this is again a relay related matter, when waiting a few seconds, everything gets cut off due to the relay breaking contact
this does not happen if not using a relay at all, im assuming because if no relay they get constant power even when "off" so they are more "ready".

id like to mention this doesnt happen when using a playlist that includes a black color - 0 brightness - 0.2s lenght preset at the start of the playlist, would be awesome if possible to make that kind of preset added across the board when using a relay, if it were added to every turning on command of wled.

closed 2023-12-31 by blazoncek