wled/WLED #3594: easy workaround to the known flicker issue with wled when using a relay

doronazl, 2023-12-17:
researched this issue a lot lately and the workaround are just a hassle to deal with to be honest.
im using a ssr to kill power to the psu when the lights are off , using WS2815, very large strips, 1000-2000led
at first i was using the relay setting on the main led menu, it works great but the problem is after the led is off, when turning the back on theres an occasional random color flicker, that only happens with the relay, searched past threads about this and i understand its a known issue that Aircookie addressed in the past.
2 solutions i could find for this 

1. make a playlist for each effect you are using containing 2 presets, 1st preset is - lights on, o brightness, black color for about 0.2s, followed by the next preset you actually want whether its an effect or solid color, then if using home assistant , direct the action to that playlist, duable but a hassle to do this for every action you want to have.

2. using multirelay usermod as it has a delay function, it wont go under 1s so had to do some modification to the code to make it into ms with the help of Frank from this group, this way worked fine, added a delay to turn on the relay first and after x amount of time send the data, no flickering issue, this is ok as long as youre not turning on wled straight into a preset, in my case its a sound reactive preset - i turn wled on , no lights, nothing, i have to switch to a solid preset first and then back to the SR preset, then it works
if turning on a solid preset, no issue works fine.
this issue wont happen if using the main relay option on led menu, but on multirelay it wont work properly for some reason- reported that as a bug as well.

what id like to request with this post is if we can have a more direct solution to this problem instead of workarounds
would be great to have an automatic delay added into the code once using the relay option under led menu or alternatively add a 200ms of black color preset into the turning on process of wled when using a relay.
i dont know how much of a trouble it is but would be great if possible.

BrickSolid28, 2024-03-24:
I’m having the same exact issue. Would love to see a delay LED signal for XX ms after relay closes option.

fcladera, 2024-04-11:
+1 to this issue. It is creating very annoying blue flashes.

bigredfrog, 2024-05-29:
+1 for a startup delay of leds

michaldz, 2025-02-25:
+1 that's really big issue