wled/WLED #3599: Unable to add new WLED device to HomeAssistant

TheLion, 2023-12-19:
### What happened?

I added a 4th WLED device to my house and network and I try to get it into HA. The previous 3 were added some time ago and all 3 were automatically discovered instantly, but the 4th one isn’t and when I try to add in manually in the WLED integration, it reports: “Unknown error occured”.

I am able to open the webgui of WLED, both on my local domain DNS name, mDNS .local and DHCP assisgned IP-address and I am able to ping the IP from my HA box.

The WLED doesn’t contain anything complex, just 10 segments and 1 preset.

### To Reproduce Bug

Until now, it has not been resolved and the same thing keeps happening

### Expected Behavior

That the new WLED device is discovered by HA, or at least that I'm able to add it manually.

### Install Method

Binary from WLED.me

### What version of WLED?

0.14.0

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Relevant log/trace output

```shell
Adding it by IP with debug logging on, produced the following info:

2023-12-17 18:22:11.161 DEBUG (MainThread) [homeassistant.components.wled] Finished fetching wled data in 0.360 seconds (success: True)

2023-12-17 18:22:14.272 DEBUG (MainThread) [homeassistant.components.wled] Finished fetching wled data in 0.291 seconds (success: True)

File "/usr/src/homeassistant/homeassistant/components/wled/config_flow.py", line 40, in async_step_user

File "/usr/src/homeassistant/homeassistant/components/wled/config_flow.py", line 117, in _async_get_device

return await wled.update()

File "/usr/local/lib/python3.11/site-packages/wled/wled.py", line 275, in update

File "/usr/local/lib/python3.11/site-packages/wled/models.py", line 694, in __init__

File "/usr/local/lib/python3.11/site-packages/wled/models.py", line 695, in <genexpr>

2023-12-17 18:22:18.602 DEBUG (MainThread) [homeassistant.components.wled] Finished fetching wled data in 2.408 seconds (success: True)

2023-12-17 18:22:20.992 DEBUG (MainThread) [homeassistant.components.wled] Finished fetching wled data in 0.191 seconds (success: True)
```
```

### Anything else?

/json/info produces:

```
{"ver":"0.14.0","vid":2310130,"leds":{"count":120,"pwr":1306,"fps":43,"maxpwr":3000,"maxseg":16,"seglc":[1,1,1,1,1,1,1,1,1,1],"lc":1,"rgbw":false,"wv":0,"cct":0},"str":false,"name":"Nanoleaf","udpport":21324,"live":false,"liveseg":-1,"lm":"","lip":"","ws":0,"fxcount":187,"palcount":71,"cpalcount":0,"maps":[{"id":0}],"wifi":{"bssid":"66:E9:31:91:62:01","rssi":-54,"signal":92,"channel":6},"fs":{"u":40,"t":1024,"pmt":1702836971},"ndc":3,"arch":"esp8266","core":"3.1.2","lwip":2,"freeheap":13792,"uptime":190465,"time":"2023-12-20, 00:08:17","opt":79,"brand":"WLED","product":"FOSS","mac":"c45bbede3cdc","ip":"192.168.206.126"}
``` 

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

TheLion, 2023-12-19:
The only explanation I can find, when I compare json/info from a working WLED device to this one, is that ws is set to 0 and on working one to 1.
Can this be the issue? How/why does this happen?

blazoncek, 2023-12-20:
Some 8266 exhibit buffer corruption due to still unknown reasons. Try reflashing by using different flash speed and mode options.

TheLion, 2023-12-20:
A reflash didn't solve the issue unfortunately. No matter what speed or mode I tried.
I did find an older WEMIS D1 Mini, in stead of the LOLIN D1 Mini and that worked better, not flawlessly, but better. Still havind some weird issues within HA

TheLion, 2023-12-23:
I am wondering. It is better/easier to use a different device than 8266?

blazoncek, 2023-12-23:
ESP8266 may be ending its life regarding WLED support.
ESP32 is the way forward in the long term.

softhack007, 2023-12-23:
> I am wondering. It is better/easier to use a different device than 8266?

Yes, classic esp32 (dual core) is the best choice for most use cases of WLED. If you have a small setup, esp32-s2 might be another option. Both are availeable in "d1 mini32" form factor.

blazoncek, 2023-12-23:
Indeed. I have replaced my D1 mini form-factor devices with Lolin ESP32-S2 D1 mini (with 2MB PSRAM) and find it excellent replacement for ESP8266. There is one caveat: it is more susceptible to WiFi signal strength or interference.

TheLion, 2023-12-23:
Thanks. Will get a Lolin ESP32-S2 D1 mini and try that. 👍

closed 2024-01-15 by blazoncek