wled/WLED #3602: wifi only works when plugged into battery bank/worklight

i3yronShepherd, 2023-12-21:
### What happened?

Stumped w/ weirdest issue - no wifi when plugged in, but works on battery ESP32-Wroom-32

I can't get my head around why this is happening...

I have WLED loaded on an ESP-Wroom-32 Dev1. Upstairs, in my tinker'n room, which is close to the router, the ESP works fine when plugged into a variety of wall warts or USB outlets, or running off the 5v light supply connected to VIN and Ground directly. Downstairs, where the lights are installed, the ESP boots, and is alive, but won't be detected, by WLED and says "offline".

BUT... if I unplug the wall adapter and plug into my worklight/battery bank, everything works fine. It goes on the WiFi network, WLED picks it up, and I can control via browser, app, etc...WTF? I can't see from a physics/electrical standpoint why this would be.

I should mention, the same usb cable is used in either scenario. The same phone chargers that work upstairs, for some reason, don't work downstairs. I tried it with and without lights, thinking the lights were drawing too much, it didn't matter - again, the ESP boots, and is active, and I can connect directly to its access point and control lights, just no WiFi capability, unless it's plugged into the battery.

I'm stumped... any ideas?

### To Reproduce Bug

I can reproduce it every time, by not plugging it into the battery

### Expected Behavior

I expected the wifi to work when plugged into another means - WLED acces point continues to work, but no wifi

### Install Method

Binary from WLED.me

### What version of WLED?

0.14.0

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2023-12-21:
Not a software but a hardware issue.
Please use WLED [forum](https://wled.discourse.group) or Discord for help and support questions.
Closing

softhack007, 2023-12-21:
This appears to be a problem related to hardware, not software. The firmware itself does not "know" what kind of power source its connected to. However some boards are known for being very sensitive to ripples in the power supply voltage (you need an oscilloscope to measure this). 

Might help to add a fat capacitor (1000 uF or more, preferably the "low ESR" kind) between Vcc and GND, to filter out ripples.

closed 2023-12-21 by softhack007