wled/WLED #3605: Enhance API to choose a random filtered effect (1D, 2D, sound reactive)

aeronauth, 2023-12-21:
It would be great if HTTP/JSON API could start a random effect with filter flags (1D, 2D, sound reactive x2) to limit the random effect to these filters.

With that, one can use a push button to randomly go trough all matrix effects only, another button to randomly go trough all sound reactive effects only, ....

As a effect might be 2d and sound reactive at the same time, that flags might be implementend for combination use.

I know that it's possible to create alot of presets and use them in playlists, which can  be used by the buttons, but i think there should be a more convenience solution.

blazoncek, 2023-12-21:
Unfortunately that may be impossible to implement without severely impacting performance as this data (1D, 2D, ...) is embeded in string. Parsing it takes time.
Best approximation is to use `"fx":"X~Y~r"` where you select X and Y to encompass certain effect range.
See [KB](https://kno.wled.ge/interfaces/json-api/#contents-of-the-segment-object)

Unfortunately I was over-voted when I tried to group certain effects together and the SR effect numbering prevailed.

dosipod, 2023-12-23:
How about adding the ability to access FX numbers directly from a playlist as it is limited at the moment to preset numbers 
[image]

dosipod, 2023-12-24:
And I just tested your command listed about for "fx":"X~Y~r" as in {"fx":"124~180~r"} or {"seg":{"fx":"124~180~r"}}  if that is what you meant and it does not work so that

blazoncek, 2023-12-24:
I am sorry for the confusion. The `"X~Yr"` API only works for presets (notice the lack of second `~`).
Unfortunately effects have hardcoded limits for random selection.
I may add custom limits in the future.

czesiu20, 2023-12-31:
> Unfortunately that may be impossible to implement without severely impacting performance as this data (1D, 2D, ...) is embeded in string. Parsing it takes time. Best approximation is to use [...]

Limiting the range of effects would be very useful. I had to implement my own function to handle Music+/- buttons of my 38-key and 24-Key Music IR remotes.

    case IR38_MUSIC_P    : changeEffect(relativeChange(effectCurrent, -1, 128, MODE_COUNT-1)); break;
    case IR38_MUSIC_N    : changeEffect(relativeChange(effectCurrent,  1, 128, MODE_COUNT-1)); break;
    
Currently it is not possible to achieve in a JSON file.

However, my solution is not completely satisfactory because I encountered another problem. Adding a new option to the list of remotes in the GUI is not trivial and I had to replace the existing one.

blazoncek, 2024-03-22:
Available in 0.15.0 beta 1.

closed 2024-03-22 by blazoncek