wled/WLED #3608: RGBW+WW support (WS2805)

Modalen-Engineering, 2023-12-22:
Hi!
Is anybody working on RGBW+WW support?
I will receive some ws2805 tape in January and see that WLED don't support it right now.
From the datasheet i can that data is similar to ws2814 with 8 extra bit for the last white.
[Worldsemi-WS2805 EN.pdf](https://github.com/Aircoookie/WLED/files/13753418/Worldsemi-WS2805.EN.pdf)

blazoncek, 2023-12-22:
No, not yet. Check if NeoPixelBus supports it.

SourceQuality, 2024-02-09:
It seems like support was added to NeoPixelBus a couple of days ago.

https://github.com/Makuna/NeoPixelBus/pull/765

https://github.com/Makuna/NeoPixelBus/issues/764

R3Cardoso, 2024-03-11:
Any news about the support about this?
I just recived it, I thought I could color it like I don on Govee M1, using the WS2814, but on this on it doesn't work. It just display random colors every group of leds.

arjanhagedoorn, 2024-03-19:
> Any news about the support about this? I just recived it, I thought I could color it like I don on Govee M1, using the WS2814, but on this on it doesn't work. It just display random colors every group of leds.

I guess this is because you need to send a extra byte. If you send 4 byte standard, it will take the byte for the next message. 

I ordered some ledstrips too, so I second this request

blazoncek, 2024-03-19:
A groundwork has been done in #3810 
If you wish to speed up, add support in `bus_wrapper.h` and other relevant files and make a PR.

blazoncek, 2024-03-24:
[image]
I will need beta testers. Please contact me on Discord in beta testing channel.

MrSpoocy, 2024-05-20:
@Modalen-Engineering which Controller you are used? Because when i read the datasheet, the Date line Need the Same voltage as the stripe (e.g 24v).

R3Cardoso, 2024-05-20:
> @Modalen-Engineering which Controller you are used? Because when i read the datasheet, the Date line Need the Same voltage as the stripe (e.g 24v).

?
Data need voltage 24v? What are you talking about. 
Also this situation is already resolved on version 0.15 betas.

Modalen-Engineering, 2024-05-21:
> @Modalen-Engineering which Controller you are used? Because when i read the datasheet, the Date line Need the Same voltage as the stripe (e.g 24v).

5v signal level. 24v supply. Same ground as supply but not same signal Level.
Controller i have used is a custom PCB with 5v linedrivers.

closed 2024-05-21 by Modalen-Engineering

harshsyd, 2024-06-08:
I have tested this, it works well and looks pretty good except every 10-30 seconds, the lights flicker to white. It stopped for 1-2 days, now it has returned. Not sure what's going on?

[image]
[image]

Happens on all effects, Solid or any other. Changing colours make no difference.

blazoncek, 2024-06-08:
HW issue.

harshsyd, 2024-06-08:
Fair, will try different ESP32 and have a look!

blazoncek, 2024-06-27:
No for that particular issue but otherwise welcome.

miauwzvisualz, 2024-08-24:
Works nice, thanks! there is no option yet for artnet with 5 colors per Pixel. That would be perfect!

mimicathz, 2024-12-06:
Hi, I made a version to control this strip with MicroPython some time ago, but I decided to switch to WLED instead of trying to compete haha.

I've been experiencing quite a few issues with all the beta versions. There's significant flicker. At first, I thought it might be due to an incorrect connection, a loose ground, or maybe even a burn micro. But when I tested my MicroPython version again, it worked perfectly.

I think it might be a timing issue. Could anyone explain me how to adjust the timings to test this?

blazoncek, 2024-12-06:
@mimicathz have a look at NeoPixelBus.