wled/WLED #3613: Drip effect crashes wled, and reboots when speed is slowed down or segments are adjusted

fribse, 2023-12-24:
### What happened?

I have a christmas tree light, and I use drip as one of the effects. It was running on esp8266, and when I adjusted the speed of the drip to run slower, it crashes the wled, and it reboots.
So I swapped it out with a esp32, as I thought the esp8266 might get overloaded (it's only 150 LEDs).
The same thing happens, as soon as I apply drip, and slow it down, it reboots the wled.

### To Reproduce Bug

Apply Drip, and slow it way down.

### Expected Behavior

Speed should be lower

### Install Method

Binary from WLED.me

### What version of WLED?

0.14.1B

### Which microcontroller/board are you seeing the problem on?

ESP8266, ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

fribse, 2023-12-24:
Same goes for popcorn, but not instantly as the drip, but after adjusting up and down on speed and intensity, it reboots

fribse, 2023-12-24:
Drip also crashes if I adjust segments.

blazoncek, 2023-12-24:
I am unable to reproduce. Tested on ESP32, ESP32-S2, ESP8266 both 1D & 2D setup.

softhack007, 2023-12-24:
I'm also unable to reproduce. I've used these effects (drip, popcorn, starburst) often, but never had any problems.

It might be an electrical problem, like
* using flimsy dupond wires
* unstable /not regulated power supply
* capacitor needed to stabilize power
* powering the LEDs directly from esp 5V
* some esp boards (especially from AZ-delivery) are very sensitive to under- or over-voltage

Check our KB for some help:
* https://kno.wled.ge/basics/getting-started/
* https://kno.wled.ge/basics/top5_mistakes/
* https://kno.wled.ge/basics/faq/

dosipod, 2023-12-24:
Might be it is better and quicker to replicate if  cfg.json is provided   whenever a crash is reported , Using WLED_0.14.1-b1_ESP32_audioreactive.bin  from gh release page I was only able to crash a unit when fps is set to 120 with effects like octopus if you change the speed ( might be the warring about setting that to max should be added same  as in MM )  but I think that might be expected at least from our testing with it on prior versions

Doyle4, 2023-12-25:
Seems same issue I [had](https://github.com/Aircoookie/WLED/issues/3609)

blazoncek, 2023-12-25:
Closing as a duplicate of #3609

closed 2023-12-25 by blazoncek

fribse, 2023-12-25:
> It might be an electrical problem, like
>     * using flimsy dupond wires
>     * unstable / not regulated power supply
>     * capacitor needed to stabilize power
>     * powering the LEDs directly from esp 5V
>     * some esp boards (especially from AZ-delivery) are very sensitive to under- or over-voltage

Always on a DigUno, amd with good PSU (5V 6A on this).

fribse, 2023-12-29:
Solved in B2