wled/WLED #3622: ESP NOW Remote presets are not fully applied

Pejosonic, 2023-12-27:
### What happened?

When switching to the four presets assigned to the buttons of a WIZ Remote they are applied but:

- Not sending to the other WLED devices using the Sync function
- If it was previusly apply a playlist, it changes to the new preset during some time but the playlist keep running (so another preset is applied after some moments).

When applying the same presets using the Web UI, everything works as expected.

### To Reproduce Bug

Use two or more WLED devices on the same network and controlling the "master" device using a WIZ Remote Control.

### Expected Behavior

To propagate the current status of master device to the remaining devices.

### Install Method

Binary from WLED.me

### What version of WLED?

0.14

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2023-12-27:
You need to enable sync on button press (settings).
Incorrect playlist handling is known and fixed in source (backported from 0.15).

Pejosonic, 2023-12-29:
> You need to enable sync on button press (settings). Incorrect playlist handling is known and fixed in source (backported from 0.15).

Thank you for these insights.

closed 2023-12-29 by Pejosonic