wled/WLED #3623: Migrated 0.13 to 0.14 presets without "Segment 0" fail to work properly

stephenhouser, 2023-12-28:
### What happened?

Updated from 0.13.3 to 0.14.1. None of my presets with segments worked. Each preset had 4 segments. Only the first 3 would light up, and none with the correct color or effect.  In addition, I could not add new segments. When I clicked add segment, then the checkbox, the newly defined segment would just "go away" as if I had never done anything.

Reverted to 0.13.3 and all worked perfectly again.

### To Reproduce Bug

Paste in a new JSON of a preset with out segment 0 defined. Then try to modify segments or presets.

```
{
	"on": true,
	"bri": 155,
	"transition": 7,
	"mainseg": 1,
	"seg": [
		{"id": 1,
			"start": 0,
			"stop": 2,
			"grp": 1,
			"spc": 0,
			"on": false,
			"bri": 255,
 			"col": [ [ 8, 255, 0 ], [ 0, 0, 0 ], [ 0, 0, 255 ] ],
			"fx": 0,
			"sx": 193,
			"ix": 255,
			"pal": 5,
			"sel": false,
			"rev": false,
			"mi": false
		},
	]
}
```

### Expected Behavior

I expected the update from 0.13 to 0.14 to migrate presets properly. If segment id=0 is now required, include checks for imported presets. Don't force a "factory reset" when Segment 0 is not defined.

### Install Method

Binary from WLED.me

### What version of WLED?

0.14

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2023-12-28:
Segment handling was rewritten from scratch and that may caused the behaviour you are seeing.
Unfortunately not much can be done except edit those presets and replace "id" field with consecutive numbers starting at 0.
You can help yoursef by using online tools like jsonformatter.org

stephenhouser, 2023-12-28:
This should be listed as a *breaking change* in the change log at least. I saw nothing in there to this effect.

blazoncek, 2023-12-28:
Well, if you'd had segments starting with 0 (as is normal and default) there wouldn't be any problems. 😄 

Unfortunately we cannot anticipate *every* possible set-up and configuration. So it didn't occur that some people would be using segment set-up without starting segment (0). Segments, in 0.13 and below, were fixed (in memory) in 0.14 (and above) are dynamic and so *need* segment 0 to exist.

If the workaround doesn't suit you please downgrade to 0.13 as 0.14 and above will not handle such configuration correctly.

stephenhouser, 2023-12-28:
I'm not asking to fix or anticipate every configuration and understand that things need to change. There should be a note or warning in the change log or **somewhere obvious** that there are **breaking changes in 0.14** if you used segments and presets. I searched again and there is nothing in there that would give anyone a clue that their 0.13 presets would not migrate properly or that they should save them before updating.

I created all the presets in the UI of 0.13, I didn't do anything outside what was allowed there. The lack of a segment with ID=0 is not something I cooked up to break things. It was made in the UI in v0.13.

softhack007, 2024-01-03:
Possibly a duplicate of https://github.com/Aircoookie/WLED/issues/2967

blazoncek, 2024-01-04:
Indeed.

closed 2024-01-04 by blazoncek