wled/WLED #3627: USERMOD BME280 HomeAssistantDiscovery - Unit Entry Missing / No Value Update / No Statistic Possible

gsieben, 2023-12-29:
### What happened?

The HomeAssistantDiscovery integration contains three bugs. 

1. The unit is not sent along with the temperatures. The field "unit_of_measurement" is missing. 
2. The values are sent at the beginning. But then the update stucks. While the values in the "wled" tree are updating as expected .
3. It is not possible to display the temperature, humidity and pressure values in the statistics. The HomeAssistant developers say that this is a bug in the data. See: https://www.home-assistant.io/more-info/statistics/

### To Reproduce Bug

Install the latest version and look into the MQTT stream (missing unit and update problem) or in HomeAssisten (statistics problem)

[image]

### Expected Behavior

1. The unit should also be transferred with the temperature. The code does contain a "tempScale" variable. But this is not transferred. I did not recognize the error immediately. 

2. The values in the HomeAssistant tree should be updated with the values of the wled tree

3. It would be nice if the statistics also worked. 

### Install Method

Self-Compiled

### What version of WLED?

WLED 0.14.1-b2

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2023-12-29:
Please contact original author as this usermod is not maintained by WLED core team.

gsieben, 2023-12-29:
OK. But - in case of a usermod problem, should I not open a new issue and mail the developer directly, or point him to this report? I did not know how to do it. Because the code itself is in this archive.

blazoncek, 2023-12-30:
We (core team) only update usermods if WLED core features change that require update in usermods. But we do not change usermod's behaviour.
Unless we (ourselves) use a particular usermod or are authors of such usermod.

You may open an issue if you want but I am trying to explain that for most usermods nothing will happen unless you contact original author (or maintainer). Please use Git commit history to find such author.

blazoncek, 2024-01-03:
Just make a PR with proposed changes and try to find other users to test and comment/report on your changes, if possible.

We only review usermod for potential conflicts with WLED functionality. We may sometimes advise on certain aspects of code and lately I have started asking authors to provide contact information for cases like this. Nothing more as usermods are just that - mods made by users to solve their requirements (which may benefit other users as well) that may not be implemented by WLED itself. And yes, errors may remain indefinitely.
The easiest way of finding who made changes to particular piece of code is Git blame. It does not mean you will get valuable contact information but at least it shows you when changes were made and by whom.

closed 2024-05-10 by github-actions[bot]