wled/WLED #3628: Presets are not shown in HomeAssistant (WLED 0.14.0)

andreas-bulling, 2023-12-30:
### What happened?

I installed the latest WLED 0.14.0 on the LED controller and added it to HomeAssistant (latest version). Everything seems to work fine except for the presets: I've added two through the WLED web interface but they are not shown in HomeAssistant. The preset list is empty.

### To Reproduce Bug

Install WLED 0.14.0, integrate into HA, add presets

### Expected Behavior

Presets are shown in HA and can be selected manually or through automations

### Install Method

Binary from WLED.me

### What version of WLED?

0.14.0

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

andreas-bulling, 2023-12-30:
Upon deleting the device in the HA WLED integration and re-adding it the presets are suddenly shown. I think I did the firmware upgrade after adding the device initially - maybe this was causing the problem...

andreas-bulling, 2023-12-30:
I have to reopen this because if I change the presets (e.g. add/remove or rename one) through the WLED web UI these changes are only visible in HA if I remove the device completely and re-add it again. This is quite cumbersome - would be great if changes automatically.

blazoncek, 2023-12-30:
This is not a WLED issue but Home Assistant issue.
WLED presets follow JSON API documented [here](https://kno.wled.ge/interfaces/json-api/).

dosipod, 2023-12-30:
@andreas-bulling Beside being related to HA wled integration , if you modify your presets in wled simply reload wled integration and they will sync  . There is no need to remove the device completely and re-add it again

andreas-bulling, 2024-01-02:
This worked - thanks a lot!

closed 2024-01-02 by andreas-bulling