wled/WLED #3630: LED Settings documentation in KB needs updating

RussellAult, 2024-01-01:
### What happened?

Some options in the LED Settings page of the UI do not currently appear in the [Settings page of the KB](https://kno.wled.ge/features/settings/#led-settings) (e.g. "Make a segment for each output", "Custom bus start indices", "Use global LED buffer", etc.). Additionally, some items in the KB appear to be in need of an update (e.g. "Skip first LED" is now the slightly different "Skip first LED*s*") or should be removed entirely (e.g. "Set current preset cycle..." doesn't appear in the UI at all).

### To Reproduce Bug

1. Open `<wled-ip>/settings/leds` in a web browser
2. Attempt to figure out what the "Use global LED buffer" setting does without searching through the project's issues/PRs 😄 

### Expected Behavior

That the options in the Settings UI would be documented somewhere (ideally in the KB, or even just in the code if it can be searched-for easily)

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.14.1-b2 (build 2312290)

### Which microcontroller/board are you seeing the problem on?

ESP32

### Anything else?

I was going to try and submit a PR with some updates, but I didn't see a way to do that for documentation.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2024-01-01:
You are welcome to update documentation if you found missing information.
Head out to [WLED-Docs](https://github.com/Aircoookie/WLED-Docs), fork and make a PR.

RussellAult, 2024-01-01:
Great! I think I've done what I can right now (it's not perfect, but I think it's better than nothing?)

Aircoookie/WLED-Docs#180

closed 2024-01-03 by blazoncek