wled/WLED #3638: Stutter/black LEDs during animation sequence with 0.14.1-b2

jambees, 2024-01-03:
### What happened?

I used the web flasher and didn't change any options and just flashed an ESP8266. It seems to have defaulted to the latest beta (0.14.1-b2) rather than the latest full release (0.14.0). I noticed with the meteor effect with the Icefire palette that every 4 sweeps of the effect across the length of my lighting strips would pause for about a half a second and there would be around 5-10 black LEDs before the effect would resume. I have put 0.14.0 firmware on and this issue has gone away. I would add that I think I saw bits of this issue with the Fireworks 1D effect but it due to the pattern it wasn't as obvious - but I'm sure there was a noticeable stutter every so often.

### To Reproduce Bug

Flash WLED v0.14.1-b2 and set the effect to meteor and palette to Icefire. Observe several complete sweeps of a long LED strip. Every 4 sweeps there'll be a black patch of 5-10 LEDs at different positions along the strip each time the problem occurs. And it is not limited to one strip as it appeared on both of my 2 strips.

My set up is a 30 LEDs/m strip consisting of 124 LEDs configured on GPIO2 and another strip of 30 LEDs/m consisting of 123 LEDs on GPIO1 with reverse checked on this strip. The meteor trail probably needs to be quite long (most of the string) to properly observe it.

I have some video footage of this happening if needed.

### Expected Behavior

No black LEDs/stutter in the effect sequence.

### Install Method

Binary from WLED.me

### What version of WLED?

0.14.1-b2

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Relevant log/trace output

```shell
None available.
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2024-01-03:
You may want to switch to ESP32. It appears that changes to WLED 0.14 overwhelm ESP8266. Especially since you are using dual output.

jambees, 2024-01-03:
I am actually intending to rebuild the hardware with ESP32 for this but I wasn't aware that 8266 was no longer really able to keep up with the latest releases.

blazoncek, 2024-01-03:
Unless you need latest features I would recommend to keep using 0.13.3 until we figure out what to strip from 0.14 to make it more viable for ESP8266.

softhack007, 2024-01-03:
> 123 LEDs on GPIO1

@jambees Please try if using GPIO3 (for the second strip) improves the situation.

> my lighting strips would pause for about a half a second

This sounds like other background activity is keeping your 8266 busy. As consequence, LEDs might freeze shortly, because the main arduino loop does not get enough time.

> Unless you need latest features I would recommend to keep using 0.13.3 until we figure out what to strip from 0.14 to make it more viable for ESP8266.

@blazoncek two shots into the wild:
* strip out websockets? I think that the UI will also work without WS?
* stripping ESP_NOW possibly helps, as ESP_NOW "competes" with WiFi for using the same radio antenna.

blazoncek, 2024-01-03:
Yes, both possible.

jambees, 2024-01-03:
> > 123 LEDs on GPIO1
> 
> @jambees Please try if using GPIO3 (for the second strip) improves the situation.

That's not going to be straightforward to try given the pcbs connecting the ESP to my lights which only caters for LED data outputs from GPIO 1&2.

Sticking with v0.14.0 is working without that issue and I'll switch to ESP32 as soon as my boards arrive - then I can retest 0.14.1-b2 or whatever the latest is at that point and see if changing hardware also resolves it.

blazoncek, 2024-01-03:
I have now tested 0.15 on my debug/development ESP8266 and I do not see any stutter with ESP8266.
It uses one digital output with 64 pixels on GPIO2 and one analog on GPIO3.
0.15 does employ a few new features and optimisations.
I do not have MQTT, Alexa, Hue enabled on that as well. It does have WS and ESP-NOW.

jambees, 2024-01-03:
In case it's relevant load wise - I do have emulate Alexa turned on, but not MQTT or Hue.

blazoncek, 2024-01-03:
Please post your configuration (cfg.json) and presets (presets.json). You can get them from backup & restore function in Security settings.

jambees, 2024-01-03:
[wled_presets_WLED-Outside.json](https://github.com/Aircoookie/WLED/files/13821062/wled_presets_WLED-Outside.json)
[wled_cfg_WLED-Outside.json](https://github.com/Aircoookie/WLED/files/13821064/wled_cfg_WLED-Outside.json)

I have no presets set up yet but have attached the file anyway. These are downloaded from the now v0.14.0 but these settings were retained during the downgrade from 0.14.1-b2 so should still be applicable.

blazoncek, 2024-01-03:
Try disabling NTP and see if stutter persists. It will prevent sunrise and sunset calculation but may help pinpointing the cause.

Doyle4, 2024-01-07:
Tried with D1 Mini ESP8266 and not happening for me, the b2 beta has been great for me so far.

blazoncek, 2024-01-15:
0.14.1 has been released. If the problem persists please reopen and update title of this issue.

closed 2024-01-15 by blazoncek

jambees, 2024-01-15:
I've just installed the release 14.1 and the same meteor with icefire effect/palette on the same ESP8266 has no stuttering now. I tried with various speeds up to max and seems fine. 👍