wled/WLED #3643: v0.14.1-b2: ESP8266 fade issue with Breathe effect on Solid and the freeze function

casesolved-co-uk, 2024-01-03:
### What happened?

See video for the interaction between the Breathe Effect and Solid

### To Reproduce Bug

Follow the video

### Expected Behavior

Solid remains solid, freeze freezes the segment and prevents changes

### Install Method

Binary from WLED.me

### What version of WLED?

v0.14.1-b2

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

casesolved-co-uk, 2024-01-03:
Related:
https://github.com/Aircoookie/WLED/pull/3631

blazoncek, 2024-01-04:
Please power your strip from a separate power supply. 1m strip with 144 LEDs/m requires 8A of current at full brightness.
There are no "interference" between segments code wise.

dosipod, 2024-01-04:
I see the same behavior when using two segments on 16x16 matrix , I do not thing this is HW related

blazoncek, 2024-01-04:
> I see the same behavior when using two segments on 16x16 matrix , I do not thing this is HW related

Please post your configuration and picture of HW set up. As mentioned above there are no interactions between different segments (apart from ABL) to do that.
ABL stands for Automatic Brightness Limiter. Turn it off and then compare.

dosipod, 2024-01-04:
I was actually testing with a remote unit using ddp but sure if you disable limiter  the issue is no longer there

softhack007, 2024-01-04:
Makes sense to explain the behaviour with ABL (auto  brightness limiter).
* segment1 is solid and power usage is close to ABL limit
* segment2 turns on, LEDs draw extra power
* ABL tries to bring power usage back into the limit, by reducing global brightness temporarily.

--> segment1 seems to fade or flicker.

This would be expected behaviour. 

@casesolved-co-uk please try this:
* verify that your cables are thick enough for 5-8Amp current
* reduce global brightness to 50% or less
* disable auto brightness limiter in LED settings
* check if the fading / flickering is still there (should be gone now)
* keep an eye on your cables, go ensure they don't heat up.

casesolved-co-uk, 2024-01-04:
@softhack007 thanks for the explanation, ABL explains the effect interaction and goes away with ABL disabled.

It cannot be a current issue in this case as the power is being fed directly to the board where the 3 LEDs are fading out. Plus the behaviour is clearly software controlled.

Current is 0.7A with ABL enabled and rises to 1.4A with it disabled, including 0.2A for the board.

And the freeze issue goes away too.

Since we're on the subject, does ABL only apply to onboard busses and is ignored for external busses, e.g. on the network?

blazoncek, 2024-01-04:
Yes.

closed 2024-01-04 by blazoncek