wled/WLED #3647: ESP32-C3 audioreactive support

cdalexndr, 2024-01-05:
[image]

Device: [ESP-C3-01M-Kit](https://docs.ai-thinker.com/_media/esp32/docs/esp-c3-01m-kit-v1.0_specification.pdf)

Related: #2735

softhack007, 2024-01-05:
This would be a request for https://github.com/Aircoookie/WLED-WebInstaller. In fact you can make a custom build for -C3 with audioreactive usermod, so its supported as much as possible already.

On the other side - honestly - you will not get happy with audioreactive on -C3. The -C3 chip is **so slow** that it cannot process audio in realtime any more. Audioreactive will use 100% CPU, and still lag behind noticeably. As a side-effect, your WLED animations will become slow, too. For -C3, my recommendation is to only use audioreactive as "sound sync receiver" client for a more capable MCU that does audio processing.

About the webInstaller: it was decided to only provide one pre-build firmware with audioreactive at this time, because audioreactive is "just" one usermod. This may change in the future, as -S3 (and -S2, if you accept some limitations) are also able to run audioreactive. 

I'm closing this report, as there is nothing planned on WLED source code side to improve audioreactive performance on -C3. Again, the -C3 chip is very slow compared to other esp32 variants, this is a limit that's very hard to "trick out".

Please feel free to make a request in request for https://github.com/Aircoookie/WLED-WebInstaller.

closed 2024-01-05 by softhack007

cdalexndr, 2024-01-06:
Thanks for response @softhack007. Do you know variants of ESP32 will run ok with audio reactive?

Maybe the page https://kno.wled.ge/advanced/audio-reactive/ should be modify to include the recommended models and a warning for other models.

cdalexndr, 2024-01-06:
Found supported models here: https://github.com/Aircoookie/WLED/blob/main/usermods/audioreactive/readme.md#supported-mcus

blazoncek, 2024-01-06:
You can update documentation and help.
Head to WLED-Docs.