wled/WLED #3651: beta versions of 14.0 cause power off strangeness on matrix display

Dramus8, 2024-01-07:
### What happened?

Pressing the power off button in the web page causes the matrix to switch to an all yellow/orange color rather than switching the LED's off.

### To Reproduce Bug

Switch matrix on to a preset display, then press the power off button in the web page.

### Expected Behavior

Expect LED panel to turn off

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 14.0.b1 and 14.0.b2

### Which microcontroller/board are you seeing the problem on?

ESP32-S3

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2024-01-07:
Unfortunately that is not the case on 30+ of my devices.
Please post your configuration and presets.

Dramus8, 2024-01-07:
[wled_cfg (1).json](https://github.com/Aircoookie/WLED/files/13854482/wled_cfg.1.json)
[wled_presets (1).json](https://github.com/Aircoookie/WLED/files/13854483/wled_presets.1.json)

I know it happens with the Jayhawk preset.

Thanks for taking a look into this for me.

blazoncek, 2024-01-07:
Please update to latest version.

Dramus8, 2024-01-07:
[image]
 Still seeing the same behavior.

blazoncek, 2024-01-08:
The issue comes from PixelArt being stored in preset.
Please use debug build with exception decoder and submit crash log.

softhack007, 2024-01-08:
> The issue comes from PixelArt being stored in preset.

Hi @blazoncek i've just tried his presets file with my own debug build - did not see any crash, but noticed something much simpler: The PixelArt preset is based on "solid". When activating the "jayhawkx" presets, pixelart is loaded from JSON. Now click on global "switch off" icon -> black. click on global "switch" again -> orange. 

I think it's simply that the pixelart JSON is not re-loaded at global switch-on, so the selected effect takes over and re-paints the matrix in orange.

blazoncek, 2024-01-08:
Thank you @softhack007 for explanation. I was dealing with two issues at the same time while being interrupted in-between. 😁 

Pixelart is a temporary display data which is lost after any change in WLED state. This is not a bug but a design choice.
There is nothing that can be done except to re-select preset.
This will not be fixed.

closed 2024-01-08 by blazoncek

softhack007, 2024-01-08:
@Dramus8 a workaround is to put your PixelArt Preset into a regularly repeating playlist.

[image]

This way the image is re-loaded every 10 seconds, so your favourite bird will re-appear after some seconds when the matrix was switched off.