wled/WLED #3653: Allow me to make zero length LED Outputs

larrywal, 2024-01-07:
**Is your feature request related to a problem? Please describe.**
I have dig-quads for light shows and have I have six placed in permanent boxes around my house.  But I mix and match props throughout the year depending on shows.  Sometimes a controller powers 2 props.  Sometimes 4.  Sometimes 1.  So the way I use various outputs change.  And when I go from 4 props to 1 or 2, I don't use the others.  Sometimes (based on how I have things wired) I also use data output 1 and data output 4 but not 2 and 3.  When I do this, I need to use null pixels on 2 and 3 because I can't have zero length outputs on them.  And when I go from 4 props to 1 and remove outputs 2 through 4, then the next time I add them I need to reset color order and GPIO.

**Describe the solution you'd like**
I'd like to be able to set 0 length LED output so I don't lose the color or GPIO settings when I'm not using the outputs.

**Describe alternatives you've considered**
Resetting things every time.

blazoncek, 2024-01-07:
Create different configurations and back them up. Restore each when needed.
You can also use /json/cfg endpoint to dynamically change configuration.

softhack007, 2024-01-08:
Currently stop-start = 0 is used as a special value in many places, to indicate stuff that is invalid or to be deleted.

Allowing busses and segments with _start=stop_ may cause lots of rework and follow-up bugs in many places of the code.

I'd suggest to keep it as-is.

closed 2024-01-12 by blazoncek

larrywal, 2024-01-13:
Bummer that 0 is used as special so this is hard.  For what it's worth, the workaround is really onerous.  I don't really consider it as such.  My configurations change all the time, so creating them, backing up, and restoring is far harder than just adjusting the values when I put up new props where I can just go to the app on my phone and set a value....

blazoncek, 2024-01-13:
You can change configuration using /json/cfg endpoint and submit your configuration on the fly.

larrywal, 2024-01-16:
Link to instructions?

blazoncek, 2024-01-16:
Source code.
You can upload (POST) entire configuration as a JSON payload to that endpoint.