wled/WLED #3655: WS2814 support

Spire9, 2024-01-07:
**Is your feature request related to a problem? Please describe.**
I bought two types of WS2814 RGBW light strips from BTF, and the WLED option for "WS281x" is wrong and does not work.  I spent a lot of time fiddling and Googling to find a workaround.  A "WS2814" option is needed.

**Describe the solution you'd like**
Add an option for setup as "WS2814" that functionally works like the settings "SK6812 RGBW"; Color Order: "BRG"; Swap: "W&G".  Additionally, the current option for "WS281x" needs to be renamed to whatever it was really designed for, "WS2811" for example.

**Describe alternatives you've considered**
I've considered not submitting this feature request to inflict my experience upon all new users.

**Additional context**
Add any other context or screenshots about the feature request here.

Thank you for your ideas for making WLED better!

blazoncek, 2024-01-08:
WS281x and SK6812 protocols are exactly the same the difference is the interpretation of channels.
Use SK6812 and swap White channel and you are done.
Won't fix.

closed 2024-01-08 by blazoncek

Spire9, 2024-01-08:
Then let's change the option's wording to say "SK6812 & WS2814 RGBW" instead of "SK6812 RGBW".

softhack007, 2024-01-09:
> Then let's change the option's wording to say "SK6812 & WS2814 RGBW" instead of "SK6812 RGBW".

Makes sense - changed in latest source code. Thanks for pointing out. 
I'll just use "/" instead of "&", because & is a trouble maker in html files.

blazoncek, 2024-01-09:
The use for WS2814 is well documented in [KB](https://kno.wled.ge/basics/compatible-led-strips/).

Spire9, 2024-01-09:
@softhack007 Awesome, you're the man!  That is a nice improvement to the new user experience.
@blazoncek I believe you that it's documented for people that have heard of WLED.  But like many others in this world I was shopping for my first addressable LEDs last month.  Search results seemed to imply there was no open-source software and controller combination that was known to support 24v WS2814.