wled/WLED #3659: Soft WDT reset crash after

CommanderRedYT, 2024-01-08:
### What happened?

After a while, my leds went out. I plugged it into my laptop to check if its brownout, but it actually was a watchdog trigger

### To Reproduce Bug

- Install the firmware from the web installer
- Connect it to WiFi
- Add it in homeassistant (i am not sure if that effects it tho)

### Expected Behavior

No crashes

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.14.0 (build 2310130)

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Relevant log/trace output

```shell
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Soft WDT reset

Exception (4):
epc1=0x4029ee72 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

>>>stack>>>

ctx: sys
sp: 3fffeda0 end: 3fffffb0 offset: 0160
3fffef00:  402c63b8 5210e673 60000600 4027b319  
3fffef10:  00000000 3ffea878 00000000 00000000  
3fffef20:  4029ec88 3ffea878 3ffefd00 60000600  
3fffef30:  402a4f85 3ffea878 3ffefd00 5210e8d3  
3fffef40:  402a4fca 3fffdab0 00000000 3fffdcb0  
3fffef50:  3ffefd18 3fffdad0 3fff37c0 402650ca  
3fffef60:  40000f49 40000f49 ffffff01 40000f49  
3fffef70:  40000e19 000d3c8f bff00000 0000bfff  
3fffef80:  00000005 aa55aa55 000000ed 40106841  
3fffef90:  40106847 bff00000 0000bfff f517b9d8  
3fffefa0:  4010000d bff00000 000d3c8f 401000ab  
3fffefb0:  402952c4 3fffef4c 40295275 3fffff68  
3fffefc0:  3fffffd0 00000000 00000000 feefeffe  
3fffefd0:  feefeffe feefeffe feefeffe feefeffe  
3fffefe0:  feefeffe feefeffe feefeffe feefeffe  
3fffeff0:  feefeffe feefeffe feefeffe feefeffe  
3ffff000:  feefeffe feefeffe feefeffe feefeffe  
3ffff010:  feefeffe feefeffe feefeffe feefeffe  
3ffff020:  feefeffe feefeffe feefeffe feefeffe  

3ffff030:  feefeffe feefeffe feefeffe feefeffe  
3ffff040:  feefeffe feefeffe feefeffe feefeffe  
3ffff050:  feefeffe feefeffe feefeffe feefeffe  
3ffff060:  feefeffe feefeffe feefeffe feefeffe  
3ffff070:  feefeffe feefeffe feefeffe feefeffe  
3ffff080:  feefeffe feefeffe feefeffe feefeffe  
3ffff090:  feefeffe feefeffe feefeffe feefeffe  
3ffff0a0:  feefeffe feefeffe feefeffe feefeffe  
3ffff0b0:  feefeffe feefeffe feefeffe feefeffe  
3ffff0c0:  feefeffe feefeffe feefeffe feefeffe  
[... 243 lines cut ...]
3ffffee0:  00000000 00004d65 00000000 4021f6dc  
3ffffef0:  3fff50de 0000001c 00000000 000007e8  

3fffff00:  00000000 00150258 0f9db22d 184ab5d2  
3fffff10:  402a683a 00000000 3fff331c 00000003  

3fffff20:  402a7a1f 3fffff50 3fff331c 4025c208  
3fffff30:  3fff425c 4026d158 2200a8c0 4025c710  
3fffff40:  402a683a 3fff657c 00150258 40243301  
3fffff50:  3ffffec0 2200a8c0 15810624 3fff37c0  
3fffff60:  00000000 00000000 3ffe8990 00000000  
3fffff70:  3fff0270 3fff37c0 402651e6 3fffefb0  
3fffff80:  3fff0270 00000000 3fff3794 40247686  
3fffff90:  00000000 001c006e 4026b198 3fff37c0  
3fffffa0:  3fffdad0 00000000 3fff3794 3fff37c0  
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2024-01-08:
Use 0.14.1-b3 and then use a debug build with exception decoder and replace crash dump.

CommanderRedYT, 2024-01-08:
@blazoncek I am not quite sure on how to do that. Where can I select debug build on install.wled.me? I can also not find it inside releases.

blazoncek, 2024-01-09:
You will need to compile yourself using MS VSC + PlatformIO and debug environment. Exception decoder is included with PIO.

softhack007, 2024-01-09:
> I am not quite sure on how to do that.

Some guidance for using the development environment is here:
* https://kno.wled.ge/advanced/compiling-wled/
* https://github.com/Aircoookie/WLED-Docs/issues/161

CommanderRedYT, 2024-01-15:
As of now I still did not have had enough time to do this.

Btw, wouldn't it make sense to let the CI also give out .elf files so you can debug it without reflashing? With the method above you will never know if it still is reproducible, as you reflash the ESP and the binary is not compiled in the same way.

blazoncek, 2024-01-15:
Using default configuration (unmodified `platformio.ini`) it will produce the binary comparable to official distribution.

softhack007, 2024-01-19:
Possibly related to
* https://github.com/Aircoookie/WLED/issues/3685

closed 2024-05-27 by github-actions[bot]