wled/WLED #3660: I want to help add the Ukrainian language

Kolan04, 2024-01-08:
### I am Ukrainian, I want to add the Ukrainian language to the application

You wrote to me in response to a review in the Google Play Market. And I want to help add the Ukrainian language to the application, but I don't know how to do it.  Tell me please.  I'm not a programmer but I want to make your app better.  How can I do this?

blazoncek, 2024-01-09:
Not possible ATM with WLED.
The translations have been requested but unfortunately texts are buried too deep and difficult to separate.

But I think you missed the repo.

Kolan04, 2024-01-10:
> Банкомат із WLED неможливий.

No, it's possible, I've already edited 95 lines

blazoncek, 2024-01-10:
I'm sorry, but WLED is not an app but a MCU firmware. Perhaps you are talking about mobile app from @Moustachauve and he indeed has an option to translate the app texts.
For WLED you'd need to change a lot of source files and that would strip English texts from WLED firmware.

Kolan04, 2024-01-10:
So I'm interested in a smartphone app.
It is possible to add an English/Ukrainian switch. That would be really
great.
https://play.google.com/store/apps/details?id=ca.cgagnier.wlednativeandroid

blazoncek, 2024-01-10:
Please consult https://github.com/Moustachauve and his "WLED Native" apps.

closed 2024-01-10 by blazoncek