wled/WLED #3661: Flickering ws2812 while using multiple stripes

Titanium177, 2024-01-09:
### What happened?

On ESP32-C3 whit WLED V0.14.1-b3 - Flickering (ws2812) while using 2x PWM + ws2812 (using just 1x PWM + ws2812 works fine)

See here: https://youtu.be/Iuy7_qMrHKg?si=UGeykAd4Z1jnEBEl

### To Reproduce Bug

Setup an ESP32-C3 whit WLED V0.14.1-b3 and connect ws2812 to IO 1,2,3 or 8. Then connect 4 mosfets to IO 4,5,6 & 7 for the PWM outputs.

5V (IO1,2,3 or 8) = ws2812
12V (IO5,6,7) = RGB Strip
24V (IO4) = W Strip

Sharing ground between every power supply.

### Expected Behavior

no flickering

### Install Method

Binary from WLED.me

### What version of WLED?

WLED V0.14.1-b3 Build 2401060

### Which microcontroller/board are you seeing the problem on?

ESP32-C3

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

softhack007, 2024-01-09:
Hi, 
the flickering in your video looks like an electrical issue - the complete ws2812 is flickering; for me it seems to lose power for very short but regular times. 

In contrast, the "software driver timing" related flickering typically shows as
a) single LEDs not showing the right color
b) short white "flashes" at random intervals (10s or more between flashes).

Unfortunately there is nothing we can do in  our software. 
The PWM driver is the standard [`ledcWrite()` from espressif framework](https://github.com/espressif/arduino-esp32/blob/c1a4055869c4062f7336e1310d369dd59af94629/cores/esp32/esp32-hal-ledc.c#L132), and the ws2812 driver is from [NeoPixelBus by makuna](https://github.com/Makuna/NeoPixelBus). 

Both drivers are not under our control.

In addition, as @blazoncek already pointed out, ESP32-C3 is still considered experimental for WLED. 
And its a lot slower (i.e. sensitive to background activities causing jitter) than classic ESP32.

duckdnssiedem, 2024-01-13:
I have the quite similar problem but with classic ESP32, one WS2815 and one PWM White. 
While WS2815 is set for solid color, after enabling PWM (even for 5% brightness), whole RBG strip changes colors randomly. 
For me is that mosfet switching causes noise on data line. 
Hint: While tested on bench this exact setup two hours earlier I didn't notice any issues, but after mounting both strips in rail problem apearred.  
Just started with that WLED thing two days ago...

Titanium177, 2024-01-25:
@softhack007 
> Hi, the flickering in your video looks like an electrical issue - the complete ws2812 is flickering; for me it seems to lose power for very short but regular times. Maybe a big capacitor will help to stabilize your power supply, [see KB](https://kno.wled.ge/basics/getting-started/#quick-start-guide).
> 

Definitely no electrical issue possible, the power supply got 40A available. Also, no voltage drop readable via Multimeter. The length of the stripes also doesn't matter.

> In contrast, the "software driver timing" related flickering typically shows as a) single LEDs not showing the right color b) short white "flashes" at random intervals (10s or more between [...]

@duckdnssiedem 
> I have the quite similar problem but with classic ESP32, one WS2815 and one PWM White. While WS2815 is set for solid color, after enabling PWM (even for 5% brightness), whole RBG strip changes [...]

I guess the mosfets are the issue which create some kind of noise. Even while it makes no sense, since i can use WS2812 + PWM RGB (also 3 mosfets in use for that) just fine. Adding then PWM W (just 1 identical mosfet) creates this flickering on WS2812 (PWM RGB doin fine). Also, changing the mosfet IO order doesn't change anything... adding PWM W creates flickering on WS2812 every time. So maybe something just with the code for PWM W?

For myself not fixable, since WLED is way too complex for my current skills. Maybe i can spot something in the code, if u tell me exactly where to look for PWM W code.

tomasz-s, 2024-01-25:
In my case it was a signal ground issue. Ground was connected only through power injection 20cm from beginning of strip, and signal ground had bad connection. 
PWM W may have different frequency than PWM RGB? Just guessing.

closed 2024-06-02 by github-actions[bot]