wled/WLED #3662: Getting list of files in the filesystem

DarthWeber, 2024-01-09:
**Is your feature request related to a problem? Please describe.**
Trying to backup all files, it is easy for well known filenames like cfg.json and presets.json - but sometimes i have additional files like patterns or mapping. They are easy to loose after flashing again.

**Describe the solution you'd like**
Maybe like /edit a command like /filelist which returns a json list of files

**Describe alternatives you've considered**
Parsing the /edit response but without luck

Thank you for your ideas for making WLED better!

dosipod, 2024-01-09:
I have no dog in this fight but i am also interested in such feature as that would open a door for a lot of automation we could do beside the clear need to backup our ledmaps as i am having hard-time doing the manual backup for those  ( scripted on some but still hard-coded ) . 
I was also testing the SD card usermod which the fs is listed on serial so might be part of the code is already their .

blazoncek, 2024-01-09:
Please use URL: `http://wled.local/edit?list=/`
This will return JSON with FS content. Replace `wled.local` with device mDNS or IP address.

closed 2024-01-09 by blazoncek