wled/WLED #3664: 160MHz on ESP8266

casesolved-co-uk, 2024-01-09:
Just a quick suggestion:

Is there a reason I can't find any reference to using ESP8266 at 160MHz?

I've done this in my `ini` file on a D1 mini and seems to work fine, but the timing of the effects might be affected possibly:

`board_build.f_cpu = 160000000L`

softhack007, 2024-01-09:
> `board_build.f_cpu = 160000000L`

fully agree with @blazoncek - also in my case (I **don't** have 8266 **older than 1 year**) it works, and of course the framerate (FPS) of some effects may be higher so they run faster.

If you're daring, you might event try to add these:
`board_build.flash_mode = dio` (normally works, some board even allow `qio`)
`board_build.f_flash = 80000000L` (only works with newest esp8266)

I'd say all this "advanced chip tuning".

Instead of tuning for an old 8266, you could also try esp32-S2 or esp32-C3.

casesolved-co-uk, 2024-01-09:
It's not overclocking, it's in the official chip specification:

[ESP8266EX](https://www.espressif.com/sites/default/files/documentation/0a-esp8266ex_datasheet_en.pdf)

casesolved-co-uk, 2024-01-11:
Any additional comments regarding 160MHz being a legitimate clock speed for all ESP8266's @blazoncek @softhack007 ? With so much going on in WLED, additional clock cycles can only help.

blazoncek, 2024-01-11:
You can use it in your own environment if you want but I am not risking the ire of thousands of users on hundreds of different configurations.
ESP8266 is at the end of its life-cycle anyway and will soon become obsolete. It has not received a serious platform update since 2 years ago.

casesolved-co-uk, 2024-01-11:
ESP8266 has another two years of life, 12 years from 2014:
[Espressif](https://www.espressif.com/en/products/longevity-commitment)

They haven't updated it because you can't improve on perfection 😄

blazoncek, 2024-03-24:
Options added in 0.14.2 and later.

closed 2024-03-24 by blazoncek

scargill, 2024-08-28:
> It's not overclocking, it's in the official chip specification:
> 
> [ESP8266EX](https://www.espressif.com/sites/default/files/documentation/0a-esp8266ex_datasheet_en.pdf)

and always ha been - may years ago I developed the now superceeded ESP-GO and I had commands to set the clock to 160Mhz.. Will interfere with some timings however.