wled/WLED #3668: ESP32 killed?

protyposis, 2024-01-10:
### What happened?

I'm running WLED on an ESP32 D1 Mini board. It's been working stable for 2 years. A few days ago, I updated to 0.14.1-b2, and today it suddenly stopped working. Now I'm wondering whether this is a hardware or software issue.

It seems like it doesn't boot up anymore, it only emits these logs:
```
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DOUT, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1084
load:0x40078000,len:11220
load:0x40080400,len:5360
entry 0x4008067c
 Ada
```

Reinstalling WLED is possible, e.g., with the web installer. It even allows me to select my Wifi network, but after confirmation, it says "Unable to connect".

Then I tried to completely reinitialize the flash memory [as documented](https://kno.wled.ge/basics/install-binary/#esp32):
```bash
esptool.py -p /dev/ttyS6 erase_flash
esptool.py -p /dev/ttyS6 write_flash 0x0 esp32_bootloader_v4.bin
esptool.py -p /dev/ttyS6 write_flash 0x10000 WLED_0.14.1-b3_ESP32_audioreactive.bin
```

<details>
  <summary>Detailed output</summary>

  ```
esptool.py -p /dev/ttyS6 erase_flash
esptool.py v3.3.3
Serial port /dev/ttyS6
Connecting......
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting.....
Detecting chip type... ESP32
Chip is ESP32-D0WD-V3 (revision v3.0)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: c0:49:ef:e6:35:64
Uploading stub...
Running stub...
Stub running...
Erasing flash (this may take a while)...
Chip erase completed successfully in 2.8s
Hard resetting via RTS pin...

esptool.py -p /dev/ttyS6 write_flash 0x0 esp32_bootloader_v4.bin
esptool.py v3.3.3
Serial port /dev/ttyS6
Connecting....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting.......
Detecting chip type... ESP32
Chip is ESP32-D0WD-V3 (revision v3.0)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: c0:49:ef:e6:35:64
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash will be erased from 0x00000000 to 0x0000ffff...
Compressed 65536 bytes to 15901...
Wrote 65536 bytes (15901 compressed) at 0x00000000 in 1.6 seconds (effective 327.8 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

 esptool.py -p /dev/ttyS6 write_flash 0x10000 WLED_0.14.1-b3_ESP32_audioreactive.bin
esptool.py v3.3.3
Serial port /dev/ttyS6
Connecting......
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting.......
Detecting chip type... ESP32
Chip is ESP32-D0WD-V3 (revision v3.0)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: c0:49:ef:e6:35:64
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash will be erased from 0x00010000 to 0x00151fff...
Compressed 1318624 bytes to 874847...
Wrote 1318624 bytes (874847 compressed) at 0x00010000 in 77.4 seconds (effective 136.3 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
  ```
</details>

It still doesn't correctly boot up, though:
```
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DOUT, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
Ada
```
(`entry` address has changed, everything else is similar)

Can anybody help me revive the board?

### To Reproduce Bug

Since this happened "randomly", without any specific user interaction, I am unable to reproduce.

### Expected Behavior

WLED boots up and runs.

### Install Method

Binary from WLED.me

### What version of WLED?

0.14.1-b2

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Doyle4, 2024-01-10:
Try using beta 3, beta worked for me with some D1 Minis, but using more than 80 leds I found beta 2 crashed the D1 mini, Beta 3 has resolved this for me.

protyposis, 2024-01-10:
I also tested beta 3 (see commands above), and 0.14.1.

softhack007, 2024-01-10:
You can also find debug builds in Srg74 firmware repository: https://github.com/srg74/WLED-wemos-shield/tree/master/resources/Firmware/%40Aircoookie/Dev/0.14.1-b3/Debug_builds

blazoncek, 2024-01-10:
FYI `Ada` stands for successful startup (initialised Serial interface for Adalight protocol).

protyposis, 2024-01-10:
Thank you very much for your quick help. I figured it out, and it was a stupid oversight.

What originally happened is that WLED became unresponsive, but the LEDs themselves were still shining. After a restart, it did not reconnect to the Wifi. After reinstalling WLED, the web installers would sometimes display the available Wifi networks (not always), but it never connected. Since all my other WLEDs worked, I assumed it must be a local error on that particular board. However, @softhack007's hint on the "WLED-AP" showed me that the board is fine, and after restarting my OpenWRT AP, everything works again. So in the end it was a Wifi problem, like in many other issues in this repo 🙈 

I guess what ultimately led me on the wrong path was the serial log. I thought that `Ada` is some kind of incomplete output (the line also contained an invalid character symbol), and was expecting an obvious "WLED boot success" message :)

closed 2024-01-10 by protyposis