wled/WLED #3679: Having less ports configured than the board supports causes configured ports to not function.

chrkov, 2024-01-13:
### What happened?

I have a 4 port WLED board made by Wasatch Pixels.   I am only using 2 of the 4 ports and I am using xlights to upload the configuration.   Because I am only using 2 ports, xlights only creates the two ports in WLED.   When the card is only configured for 2 ports, nothing works on those two ports.  I cant control any lights.   If I manually add the 2 additional ports as just a single led each, then everything will work correctly.   I can also manually configure the system for 2 ports and get the same behavior, so its not an xlights thing,  It doesnt make a difference what I add the 2 ports as, can be ws2811 or even DDP or ArtNet.  Seems because its 4 ports, it has to have 4 ports configured to work.

### To Reproduce Bug

Create 1, 2 or 3 strings on a 4 port system.

### Expected Behavior

I would expect if only 2 ports of 4 were configured, that the 2 ports would work normally.

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.14.1-b3

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2024-01-13:
Please consult xLights if the correctly configure your board.

chrkov, 2024-01-14:
Its not xlights.   If I manually configure the WLED board, it does that same thing.   Xlights is configuring it correctly, its just not working with by xlights upload or by manually configuring it for only using 2 ports.

blazoncek, 2024-01-14:
Then your configuration is wrong. WLED is functioning properly with multiple outputs configured.
Please consult your hardware manufacturer for correct configuration.

chrkov, 2024-01-14:
Can you explain to me how in this configuration Ports 1 and 2 work perfect:
[image]

But in this configuration, they do not function at all?
[image]

Ports 1 and 2 are configured exactly the same.  Only difference is there is no output 3 and 4 configured.
This makes no sense of how its a configuration issue when the configuration for outputs 3 and 4 have nothing to do with 1 and 2.

blazoncek, 2024-01-14:
Please use WLED [forum](https://wled.discourse.group) or Discord for help and support questions or contact manufacturer of the controller.

closed 2024-05-23 by github-actions[bot]