wled/WLED #3680: Restore state to OFF causes LED freeze

TheHolyRoger, 2024-01-14:
### What happened?

When Hyperion disconnects with restore state enabled and the previous state being OFF, my LED strips no longer respond to any commands sent by WLED until a WLED reboot (no need to cut power to the strip or esp32) seems to be purely a software issue.

Happening on the latest release using binaries and self compiled on 3 different boards.

No other issues, working perfectly fine and smooth just if I restore state to OFF I have to reboot my boards to be able to use the lights 

### To Reproduce Bug

Restore state to OFF

### Expected Behavior

Should still respond

### Install Method

Binary from WLED.me

### What version of WLED?

0.14.1-b3

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2024-01-14:
Not a bug in WLED.
When Hyperion starts it instructs WLED to enter infinite realtime mode. When Hyperion exits it should instruct WLED to exit such mode. If the Hyperion stops incorrectly then no such info is sent to WLED and WLED will wait until you manually override such mode or restart WLED device.

TheHolyRoger, 2024-01-14:
@blazoncek WLED is exiting the mode, i log into the interface and i can control it without the "override" message, but the led strip is not responsive until WLED is restarted

blazoncek, 2024-01-14:
Please post a debug log and reopen when done.
Get a debug build and monitor serial interface.

Aircoookie, 2024-01-14:
Just a hunch, but WLED destroys busses when off so that the builtin LED of some boards turns off. Might be that if turned off via exiting Hyperion, this might not properly recreate busses when turning on.

TheHolyRoger, 2024-01-14:
@blazoncek thanks I'll play with it tomorrow and get the log output.

@Aircoookie is there a way I can test this or change something? I haven't played with the bus or global buffer settings from default

TheHolyRoger, 2024-01-14:
Just to be clear, if I restore state to an ON state, everything is working fine, just not if the previous state was OFF

blazoncek, 2024-01-14:
Check if your segments are "frozen" ("eye" icon in segment).

TheHolyRoger, 2024-01-14:
@blazoncek nope not frozen, it was displaying the clock icon as usual, i tried toggling it to eye and back with no response from the LED strip just now, toggled it on/off, changed effects etc but still nothing until i rebooted WLED.

Will still try and get some logs tomorrow with fresh builds with debug enabled (this issue is present for me on my current custom builds and pre-compiled binaries from wled)

TheHolyRoger, 2024-01-29:
@blazoncek I'm not seeing a lot of useful info in the debug log when this happens, not sure if I'm missing some compile options? Only one error right at the end regarding locking json buffer

```
Not-Found HTTP call:
URI: /presets.json
WS FileRead: /presets.json
Loop took 90ms.
Usermods took 0ms.
Strip took 0ms.
JSON buffer locked. (17)
JSON buffer size: 2384 for request: 3
JSON content length: 1316
JSON buffer released. (17)
Not-Found HTTP call:
URI: /presets.json
WS FileRead: /presets.json
Loop took 90ms.
Usermods took 0ms.
Strip took 0ms.
JSON buffer locked. (17)
JSON buffer size: 2384 for request: 3
JSON content length: 1316
JSON buffer released. (17)
Local time: 16:08
---DEBUG INFO---
Runtime: 58082635
Unix time: 58082,635
Free heap: 204232
Wifi state: 3
State time: 0
NTP last sync: 999000000
Client IP: XXX.XXX.XXX.XXX
Loops/sec: 2820
Loop time[ms]: 669/90
UM time[ms]: 0/1
Strip time[ms]: 0/1
Segments: 1 -> 68B
Modes: 4*187=748B
Data: 4*187=748B
Map: 2*0=0B
Buffer: 3*138=414B
Not-Found HTTP call:
URI: /presets.json
WS FileRead: /presets.json
Loop took 89ms.
Usermods took 87ms.
Strip took 0ms.
JSON buffer locked. (17)
JSON buffer size: 2384 for request: 3
JSON content length: 1316
JSON buffer released. (17)
Not-Found HTTP call:
URI: /presets.json
WS FileRead: /presets.json
Loop took 91ms.
Usermods took 0ms.
Strip took 0ms.
JSON buffer locked. (17)
JSON buffer size: 2384 for request: 3
JSON content length: 1316
JSON buffer released. (17)
Not-Found HTTP call:
URI: /presets.json
WS FileRead: /presets.json
Loop took 90ms.
Usermods took 0ms.
Strip took 0ms.
JSON buffer locked. (17)
JSON buffer size: 2384 for request: 3
JSON content length: 1316
[... 255 lines cut ...]
URI: /presets.json
WS FileRead: /presets.json
JSON buffer locked. (14)
JSON buffer released. (14)
JSON buffer locked. (17)
JSON buffer size: 849 for request: 1
JSON content length: 483
Loop took 92ms.
Usermods took 0ms.
Strip took 3ms.
JSON buffer released. (17)
JSON buffer locked. (17)
JSON buffer size: 2368 for request: 3
JSON content length: 1303
ERROR: Locking JSON buffer failed! (17)
JSON buffer released. (17)
JSON buffer locked. (12)
JSON buffer size: 2348 for WS request (1303).
heap 201052
Sending WS data to multiple clients.
JSON buffer released. (12)
Loop took 127ms.
Usermods took 0ms.
Strip took 2ms.
```

blazoncek, 2024-01-29:
> @blazoncek I'm not seeing a lot of useful info in the debug log when this happens, not sure if I'm missing some compile options? Only one error right at the end regarding locking json buffer

On the contrary there is a very useful info there. Your loop is taking an awful lot time to complete. Normally that should be in a matter of couple of ms, not in the range of 100ms.

There's also a periodic request for state and info (about every 10s, from which I'd suspect HomeAssistant integration). Try to disconnect that and see if anything changes.

Error while locking only means that two things tried to use JSON buffer at the same time (perhaps a UI and some other task). This is controlled so no issues arise from that (on WLED backend side).

TheHolyRoger, 2024-01-29:
@blazoncek brilliant thanks I'll try disabling hass later and see if it behaves better.

What would cause my loop to take so long? would that be hass too?

TheHolyRoger, 2024-01-29:
@blazoncek @Aircoookie I opened a PR that fixes this issue for me:
https://github.com/Aircoookie/WLED/pull/3731

TheHolyRoger, 2024-01-29:
Fixed in https://github.com/Aircoookie/WLED/commit/220217561a361f3ba49aa9b0bbe97375396d816c

closed 2024-01-29 by TheHolyRoger