wled/WLED #3686: ESP8266 websockets causing crash

casesolved-co-uk, 2024-01-15:
### What happened?

Clicking lots of times on the colour wheel causes a crash whilst DMX is outputting.
I'm using my own DMX driver but anything sending effect data over a HardwareSerial driver should be sufficient to cause it.

The WDT reset below doesn’t happen if I use WLED_DISABLE_WEBSOCKETS

Here’s the reset output:

```
heap 14992
Sending WS data to multiple clients.
JSON buffer released. (12)

 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x4010f000, len 3424, room 16 
tail 0
chksum 0x2e
load 0x3fff20b8, len 40, room 8 
tail 0
chksum 0x2b
csum 0x2b
v000c7a00
~ld
����'�;��'|�$�$ c��|;�l�g�
                          �o�l`��s�l�l
                                      �
---WLED 0.14.1 2401141 INIT---
```

### To Reproduce Bug

Clicking lots of times on the colour wheel causes a crash whilst DMX is outputting.
I'm using my own DMX driver but anything sending effect data over a HardwareSerial driver should be sufficient to cause it.

### Expected Behavior

I can click anything as much as I want with WLED_DISABLE_WEBSOCKETS and no crash

### Install Method

Self-Compiled

### What version of WLED?

WLED 0.14.1 2401141

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Relevant log/trace output

```shell
heap 14992
Sending WS data to multiple clients.
JSON buffer released. (12)

 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x4010f000, len 3424, room 16 
tail 0
chksum 0x2e
load 0x3fff20b8, len 40, room 8 
tail 0
chksum 0x2b
csum 0x2b
v000c7a00
~ld
����'�;��'|�$�$ c��|;�l�g�
                          �o�l`��s�l�l
                                      �
---WLED 0.14.1 2401141 INIT---
```
```

### Anything else?

Sounds very similar to #3609
Related: https://wled.discourse.group/t/board-wdt-reset-caused-by-websockets/10770

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2024-01-15:
Probably RAM depletion. When you disable websockets you free up plenty of RAM.
Please upgrade to ESP32.

casesolved-co-uk, 2024-01-15:
> Probably RAM depletion. When you disable websockets you free up plenty of RAM. Please upgrade to ESP32.

Arguably it's worse and more laggy with the following disabled:

```
  -D WLED_DISABLE_LOXONE
  -D WLED_DISABLE_ADALIGHT
  -D WLED_DISABLE_MQTT
  -D WLED_DISABLE_HUESYNC
```

There's definitely an issue with websockets. Normal HTTP requests are much more responsive and reliable.

blazoncek, 2024-01-15:
I use websockets without issues on about 10 of my ESP01 devices.
I did disable 2D, Loxone, Hue, Ada, Alexa and MQTT. I also used `WLED_USE_UNREAL_MATH` to free a few bytes more by removing libc dependency..

blazoncek, 2024-01-15:
And BTW I am unfamiliar with `0x4021bea1 in BusDMX::setPixelColor(unsigned short, unsigned int) at ??:?`

This appears to be your extension of WLED which you'll need to debug on your own. I am closing as it appears not to be related with WLED core functionality.

Feel free to reopen when you reproduce with stock WLED.

closed 2024-01-15 by blazoncek

casesolved-co-uk, 2024-01-15:
It's clearly not a RAM issue because there's 14k of heap remaining when it crashes.
No issue with websockets disabled, so it can't be my driver.

Similar issues with `WLED_USE_UNREAL_MATH`

Two other symptoms:
```
heap 17664
heap 15904
Sending WS data to multiple clients.
JSON buffer released. (12)

User exception (panic/abort/assert)
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Abort called

>>>stack>>>

ctx: cont
sp: 3ffffed0 end: 3fffffd0 offset: 0010
3ffffee0:  00000001 0000b5e3 3fff4e40 4010169b  
3ffffef0:  3ffe9c96 0000b5e3 3fff4e40 40215531  
3fffff00:  00000066 0000b5e3 3fff2950 402170d4  
3fffff10:  00000000 0000b5e3 26e978d4 00d24ea4  
3fffff20:  00000000 3fff33c8 40254ac2 3fff33c8  
3fffff30:  3fffdad0 00000000 0000b5e3 4022c154  
3fffff40:  3fff4ec8 3fff29ac 3ffe8940 00000000  
3fffff50:  3fffdad0 3fff33c8 40254ac2 3fff33c8  
3fffff60:  3fffdad0 00000000 0000b5e3 40238c9b  
3fffff70:  4025c324 ffffffff 00000080 402564e8  
3fffff80:  00000000 fffffffc 0000b5d8 3ffefff0  
3fffff90:  4023f93c 3ffe9c94 3fff31b4 3fff33c8  
3fffffa0:  3fffdad0 00000000 3fff339c 3fff33c8  
3fffffb0:  3fffdad0 00000000 3fff339c 40254c50  
3fffffc0:  feefeffe feefeffe 3fffdab0 40100a01  
<<<stack<<<

0x4010169b in free at ??:?
0x40215531 in Segment::stopTransition() at ??:?
0x402170d4 in WS2812FX::service() at ??:?
0x40254ac2 in yield at ??:?
0x4022c154 in handleNightlight() at ??:?
0x40254ac2 in yield at ??:?
0x40238c9b in WLED::loop() at ??:?
0x4025c324 in StreamNull::~StreamNull() at ??:?
0x402564e8 in uart_write at ??:?
0x4023f93c in HardwareSerial::write(unsigned char const*, unsigned int) at ??:?
0x40254c50 in loop_wrapper() at core_esp8266_main.cpp:?
0x40100a01 in cont_wrapper at ??:?

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 3424, room 16 
tail 0
chksum 0x2e
load 0x3fff20b8, len 40, room 8 
tail 0
chksum 0x2b
csum 0x2b
v000bf290
~ld
���'�;��g|�
           $
            �
             d`#�|s�d�'��o�
                           d`��s�l�l
                                    �
---WLED 0.14.1 2401141 INIT---
esp8266 3.1.2
heap 30488
PIN ALLOC: Pin 1 successfully allocated by 0x89 (137)
Registering usermods ...
heap 30488
Reading config
```

```
JSON buffer released. (11)
JSON buffer locked. (12)
JSON buffer size: 3049 for WS request (1707).
heap 17640
heap 15888
Sending WS data t**ERROR: Locking JSON buffer failed! (12)**
ERROR: Locking JSON buffer failed! (12)
o multiple clients.
JSON buffer released. (12)
Loop took **2026ms**.
Usermods took 0ms.
Strip took 1ms.
JSON buffer locked. (11)
JSON buffer released. (11)
```

There's clearly a huge issue with websockets ...

blazoncek, 2024-01-15:
Websockets were not developed by WLED team. Please open an issue with AsyncWebServer/AsyncTCP library.

casesolved-co-uk, 2024-01-15:
I'll just disable it thanks. It clearly has some serious issues that I don't have time to debug. My board has been running for 18 hours outputting DMX using Candle Multi without issues with it disabled with web clicks any time. It crashes within 30 seconds with it enabled and clicks.

Websockets is a chosen feature of the project you support.