wled/WLED #3688: Add the Languages tab. I will help with the translation into Ukrainian

Kolan04, 2024-01-15:
I will help with the translation into Ukrainian for free.

I would like you to add a tab with the choice of interface language.

If it needs to be fixed in not too many files, I can help with translation into various popular languages also for free.

I bought a lamp, and the lamp manufacturer uses your interface and it would be convenient to have the Ukrainian language.

Best regards,
  Mykola Popovych from Ukraine
[image]

softhack007, 2024-01-15:
Basically the same question from here:
* https://github.com/Aircoookie/WLED/issues/3660

@Kolan04 WLED today cannot support several languages, and we have no possibility to store translations without "deleting" the original English UI - It's not just about "adding a tab".
* Flash space is very limited. We are already "on the edge" with esp8266, and esp32 firmware size is becoming a problem, too. Storing HTML UI in two languages in flash would not be possible right now. 
* Our current software design cannot manage several languages. It will be a major programming effort to achieve it.

We value your effort and your generous offer to invest time into WLED. However right now we could not integrate your translation. A lot of software design work  (programming) would be needed first - and I'm sorry to say, this is not possible in the next future.

Kolan04, 2024-01-15:
Got it, thanks for the reply, I thought I wrote to someone else because
there was an update in the app, and I followed the link and got to you
again, I thought it was different interfaces and I was writing to another
person. You have a great application, thank you for being there, I will
understand in English, because this is a web interface, I entered through
Google Chrome and there is an automatic translation of pages.

Thank you for the answer, then I consider this question closed, I will not
write to you again)))

softhack007, 2024-01-15:
@Kolan04 it might be an option to integrate automatic translation - like google chrome can do - into the Phone App. This would be completely Independent from the "WLED core", as its a feature of the browser. 

 @Moustachauve might be interested to integrate such a feature (i.e "google chrome" auto-translations) into his app.

* https://github.com/Moustachauve/WLED-Native-Android

sudofile, 2025-04-20:
@Kolan04 How did you change the language?